rokath/trice #150: Add param count check

rokath, 2021-09-24:
When during `trice u` a trice statement  ist evaluated,  the format specifier count  should  match the  param count and **_n** if specified. If not report an error and write *#error* in Front of trice statement. This allows generally  not to use **_n**.

th-sabik, 2021-11-01:
A runtime check exists now, but a pre-compiletime check (`trice -u`) is higher quality

rokath, 2022-05-21:
Merged with #291

rokath, 2023-05-11:
For example this line caused a problonly at runtime:
```C
trice( iD( 1278), "diag:RTT0_bytesInBufferMax=%u, RTT0_writeSpaceMin\n", RTT0_bytesInBufferMax, RTT0_writeSpaceMin );
```

rokath, 2024-08-17:
Write tests for `trice i -addParamCount` and `trice u -addParamCount` for TRICE, TRICE8,... ,trice64.

rokath, 2024-08-23:
parameter count check is implemented in v0.67.0

closed 2024-08-23 by rokath