rokath/trice #242: Add target context option

rokath, 2022-01-06:
This could be a separate 32-bit entry but for efficiency it could be part of the target location, for example:
- 8 bit for task context allowing 256 tasks
- 14 bit for line number, allowing 16000 lines per file
- 10 bit for file ID, allowing 1000 project files

rokath, 2022-01-07:
It is very simple to implement, nearly free of runtime costs and allows users to see the task ID alongside the source code, when usind *Trice* with a multitasker.

th-sabik, 2022-01-16:
Possible steps:
- Target code:
  - `#define id(n) ((uint32_t)n<<22) //!< id() is a 10 bit id 0-1023`
  - `#define TRICE_FILE Id(7777)` -> `#define TRICE_FILE id(7777)`
  - `#define TRICE_LINE (__LINE__<<8)`
  - `#define TRICE_CONTEXT 255 //!< any 8-bit variable will do`
  - `#define TRICE_LOCATION(TRICE_FILE| TRICE_LINE | TRICE_CONTEXT )`
  - `#define TRICE_LOCATION2(TRICE_LOCATION )`
  - `#ifdef TRICE_LOCATION2` -> TRICE_COBS_PACKAGE_MODE +=4
- Host code:
  - Assign only IDs 1-1023 to id(0)
  - Handle COBS Package mode 4-7 accordingly

rokath, 2022-05-15:
See #266 as future solution.

closed 2022-05-15 by rokath