rokath/trice #245: Feature request: receive Trice data stream from IP:port

dsvf, 2022-01-17:
Feature request: add the option to receive a Trice data stream from an IP:port combination.

My use case, where this would be very helpful: I use OpenOCD and its built-in RTT feature. OpenOCD then starts a server on localhost:17001 where it dumps all RTT messages. I can connect to this IP:port with telnet to receive these messages.
Unfortunately, that means that I cannot use the J-Link or STlink "ports" in Trice.
It would be great if I could instruct Trice to connect to this port and decode the data from that stream.

a useful side effect: this would allow network-connected microcontrollers (e.g. the ESP8266) to stream Trice messages to a host decoder without requiring a direct debugger/UART connection.

rokath, 2022-01-19:
That is a good  idea. To implement  it I first need a device sending *Trice* messages over  TCP/IP for testing. This could do a PC program written in C. Or, I add a routing mode to the **trice** tool transferring received *Trices* to TCP/IP. Or dump a test set and use it for tests. Because  *Trice* is a spare time  project  please be patient. 

Workaround option: The RTT to **trice** tool implementation uses a temporary file as interface between the JLinkLogger and the **trice** tool. So if telnet would write the received data stream in a file, the **trice** tool just needs to expose its file interface.

rokath, 2022-02-04:
Add OpenOCD  use case.

dsvf, 2022-02-05:
Concerning the testing: I have in the past used netcat or ncat for this (under Linux, ncat supposedly also works under Windows). You can use it to dump files or text over a TCP/IP connection. The simplest use case would be 
`cat file_with_trice_data_can_be_binary.txt | nc -l localhost 12345`. If you then connect to localhost, post 12345, you receive that file as a stream of bytes.

I guess piping the output of telnet into a temporary file should also work (except that telnet prints some header text that trice would have to ignore)

th-sabik, 2022-02-06:
Fixed in pre v0.48.0: Pushed to main but not released yet. Needs more testing. Seee also updated RTT document.
```
  -args string
   ...
        port "TCP4": default="localhost:17001", use any IP:port endpoint like "127.0.0.1:19021"
...
 -port string
        receiver device: 'BUFFER|DUMP|FILE|JLINK|STLINK|TCP4|serial name.
```

closed 2022-02-06 by th-sabik

rokath, 2022-02-08:
Thanks for the testing hint. I could test with the jlink binary which starts a server at localhost:19021. Please let me know if you succeded with OpenOCD and *Trice*.

escherstair, 2022-05-19:
Hi @dsvf 
in your first message you wrote
> a useful side effect: this would allow network-connected microcontrollers (e.g. the ESP8266) to stream Trice messages to a host decoder without requiring a direct debugger/UART connection.
Did you mean that the microcontroller can directly stream Trice mesage over a native TCP interface?
This is an useful scenario for me, and I don't understand if this is possible with the latest Trice realease (0.51.0) or not.

th-sabik, 2022-05-19:
The target trice code expects a provided Write() function. Example from triceConfig.h:
```C
///////////////////////////////////////////////////////////////////////////////
// RTT interface: Adapt to your device.
//

#ifdef TRICE_RTT_CHANNEL
#include "SEGGER_RTT.h"

#define TRICE_WRITE( buf, len ) do{ \
    SEGGER_RTT_Write(TRICE_RTT_CHANNEL, buf, len ); \
    ToggleOpticalFeedbackLED(); \
}while(0)
#endif // #ifdef TRICE_RTT_CHANNEL
```
If your target can write to TCP you simply use its write function. The **trice** tool can receive then.