rokath/trice #248: Call Stack Logger – Function instrumentation as a way to trace program’s flow of execution - blogersii

th-sabik, 2022-01-19:
https://sii.pl/blog/call-stack-logger-function-instrumentation-as-a-way-to-trace-programs-flow-of-execution/

rokath, 2022-01-20:
This could be an application example for gcc compiled projects.

rokath, 2022-05-22:
Now part of similar projects list

closed 2022-05-22 by rokath

TomaszAugustyn, 2022-08-16:
Hello @rokath / Thomas

I checked your project and I must say it looks incredible. Although I'm not working with C, I can see how much hard work you have put into this project. It's very sophisticated, great work!

I have also noticed that you put my project (Call Stack Logger) into "Similar Projects" section and I'm honored and grateful for that. I noticed that Sii has recently changed their blog appearance and the code snippets look very bad there. That's why I migrated the article to Dev.to and Hashnode. Maybe you could update the link to the article? 
https://dev.to/taugustyn/call-stack-logger-function-instrumentation-as-a-way-to-trace-programs-flow-of-execution-419a

Best regards,
Tomasz.

rokath, 2022-08-17:
Thanks for the update, Tomaz, and your * :-)

Thomas