rokath/trice #251: trice update changes line endings of entire file

liborw, 2022-01-28:
**Describe the bug**
trice update changes line endings of entire file

**To Reproduce**
Steps to reproduce the behavior:

```sh
$ file Core/Src/main.c 
Core/Src/main.c: C source, ASCII text, with CRLF line terminators
$ trice update
Read ID List file /home/libor/Personal/learn/trice/til.json with 11 items.
#define TRICE_FILE Id(34120) inserted into main.c
$ file Core/Src/main.c
Core/Src/main.c: C source, ASCII text, with CRLF, LF line terminators
```

**Expected behavior**
The line endings will stay the same.

**Desktop (please complete the following information):**

- OS: archlinux
- Version: latest master (trice-git v0.18.4.r703.g8b91772b-3)

rokath, 2022-01-29:
This is as expected. The **trice** tool reads files line ending agnostic, but writes files according to the OS. I guess a vice versa behaviour you described in Windows. If this is a real issue, please open the issue again.

closed 2022-01-29 by rokath

liborw, 2022-02-17:
Actually the problem is that CubeMX generates windows line endings on whatever OS. It might be better to write same line ends as it reads...

rokath, 2022-04-02:
If you prefer to work with Windows line endings in Linux, you could use dos2linux before and linux2dos after  you activated `trice update` in a script. Other option: use dos2linux after CubeMX code generator.