rokath/trice #268: Go build fails

kraiskil, 2022-03-26:
With latest master branch, the 'go' build fails:

```
$ go vet
go: finding github.com/dim13/cobs v1.0.0
go: finding github.com/mattn/go-runewidth v0.0.7
go: finding github.com/gdamore/encoding v1.0.0
go: finding github.com/lucasb-eyer/go-colorful v1.0.3
go: github.com/dim13/cobs@v1.0.0: unknown revision v1.0.0
go: finding golang.org/x/sys v0.0.0-20200116001909-b77594299b42
go: finding golang.org/x/crypto v0.0.0-20190308221718-c2843e01d9a2
go: finding golang.org/x/sys v0.0.0-20190626150813-e07cf5db2756
go: finding golang.org/x/text v0.3.0
go: finding golang.org/x/tools v0.0.0-20180917221912-90fa682c2a6e
go: finding gopkg.in/check.v1 v0.0.0-20161208181325-20d25e280405
go: finding gopkg.in/yaml.v2 v2.2.2
go: error loading module requirements
```

Or am I doing something incorrectly?
This is on Ubuntu 20.04, `go version go1.12.2 gccgo (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0 linux/amd64`

rokath, 2022-03-27:
Did you enter `go vet` or `vo vet ./...`?

kraiskil, 2022-03-27:
I got the same result for both, and I copied only the latter here.

But switching the go compiler to `go version go1.13.8 linux/amd64` made it nudge forward.
Now the build fails with:
```
$ go vet ./...
[snip: a lot of downloading/extracting logs]
go: finding github.com/dim13/cobs v1.0.0
# github.com/rokath/trice/pkg/tcobs
cannot load DWARF output from $WORK/b215//_cgo_.o: decoding dwarf section info at offset 0x0: too short
# github.com/rokath/trice/internal/decoder
internal/decoder/cobsDecoder.go:473:12: possible misuse of unsafe.Pointer
internal/decoder/cobsDecoder.go:489:12: possible misuse of unsafe.Pointer
internal/decoder/cobsDecoder.go:507:12: possible misuse of unsafe.Pointer
internal/decoder/cobsDecoder.go:525:12: possible misuse of unsafe.Pointer
```
Go version 1.13 is the newest one this Ubuntu provides. Maybe I should try with the latest one from the Go website...?

kraiskil, 2022-03-27:
Go version 1.18 (and me learning about where go puts built items...) fixes this. Thank you for the tip.

closed 2022-03-27 by kraiskil

rokath, 2022-03-27:
That's fine! The warning I know.  It is not dangerous. Right now no idea how to get rid of it.