rokath/trice #274: invalid operands to binary

ypearson, 2022-04-27:
trice.h:339:19: error: invalid operands to binary | (have 'char *' and 'int')
  339 |     TRICE_PUT( id | 0x0000 | TRICE_CYCLE ); \

I"m trying to try this on a demo project. I get this error when trying to print using TRICE

rokath, 2022-05-02:
Please do not use TRICE_PUT. This is an internal macro. Use TRICE as a printf replacement.

ypearson, 2022-05-02:
I'm not, I'm just using the example provided:
`TRICE( Id(60363), "1/11 = %g\n", aFloat( 1.0/11 ) );`

ypearson, 2022-05-02:
Here is the whole trace
/home/docker/host/External/trice-0.48.0/pkg/src/trice.h:339:19: error: invalid operands to binary | (have 'char *' and 'int')
  339 |     TRICE_PUT( id | 0x0000 | TRICE_CYCLE ); \
      |                   ^
/home/docker/host/External/trice-0.48.0/pkg/src/trice.h:122:56: note: in definition of macro 'TRICE_PUT'
  122 | #define TRICE_PUT(x) do{ *TriceBufferWritePosition++ = x; }while(0) //! PUT copies a 32 bit x into the TRICE buffer.
      |                                                        ^
/home/docker/host/External/trice-0.48.0/pkg/src/trice.h:284:18: note: in expansion of macro 'TRICE0'
  284 | #define TRICE_0  TRICE0  //!< Only the format string without parameter values.
      |                  ^~~~~~
/home/docker/host/External/trice-0.48.0/pkg/src/trice.h:228:22: note: in expansion of macro 'TRICE_0'
  228 | #define CONCAT(a, b) a ## b
      |                      ^
/home/docker/host/External/trice-0.48.0/pkg/src/trice.h:231:23: note: in expansion of macro 'CONCAT'
  231 | #define CONCAT2(a, b) CONCAT(a, b)
      |                       ^~~~~~
/home/docker/host/External/trice-0.48.0/pkg/src/trice.h:234:29: note: in expansion of macro 'CONCAT2'
  234 | #define TRICE(id, fmt, ...) CONCAT2(TRICE_, COUNT_ARGUMENTS(__VA_ARGS__))(id, fmt, ##__VA_ARGS__)
      |                             ^~~~~~~
/home/docker/host/Core/Src/main.c:107:7: note: in expansion of macro 'TRICE'
  107 |       TRICE( "1/11\n", aFloat( 1.0/11 ) );

rokath, 2022-05-05:
Hope to find time next week fort this.

rokath, 2022-05-05:
Do you use v0.48.0 or v0.49.1?

rokath, 2022-05-05:
Please use v.48.0 for now.

ypearson, 2022-05-06:
trice_0.48.0_Linux_x86_64

rokath, 2022-05-14:
Please check that in trice.h around line 220 (The comment was updated after v0.49.1) ans see also #279:
```C
//! COUNT_ARGUMENTS builds upon NTH_ARGUMENT. The more arguments that are passed to COUNT_ARGUMENTS, 
//! the more the »counting arguments« (12, 11, 10, 9, 8, 7…) are pushed to the right. 
//! Thus the macro evaluates to the number of arguments that are passed to the macro.
//! If you set the C language to strict C (C90, C99, C11 or C17) the `##` operator doesn't remove the comma before it when `__VA_ARGS__` expands to nothing.
//! In this case, the TRICE macro doesn't work with no parameters. You must explicitly use TRICE0 instead of TRICE for a no parameter value TRICE in this case.
//! For more details see closed Issue #279. Special thanks @escherstair.
#define COUNT_ARGUMENTS(...) NTH_ARGUMENT(dummy, ## __VA_ARGS__, 12, 11, 10, 9, 8, 7, 6, 5, 4, 3, 2, 1, 0)
```
If this doea not solve your issue, please provide me your Linux demo project, that I can try to get into it.

rokath, 2022-06-06:
Seams to be invalid. Please reopen just in case.

closed 2022-06-06 by rokath