rokath/trice #275: gcc/opencm3 regression

kraiskil, 2022-04-30:
**Describe the bug**
GCC/libopencm3 build does not work with latest master (7d9c7761c8e5f86a4c0ffc57391dec6f546318e8).
Host side 'trice log -p /dev/ttyACM0' does not print anything. The target application seems to be running just fine without problems,
only the Trice-prints are missing.

**To Reproduce**
Compile and run the `test/OpenCM3_STM32F411_DISCOVERY` sample - using direct printing 'mode 0'.

**Desktop (please complete the following information):**
Linux, Ubuntu 20.04, arm-none-eabi-gcc-9.3.1.

**Additional context**

I git bisected this regression from the working merge of the OpenCM3 example to the commit:
```
$ git bisect good
1a2d60afe17bbcc43dfd388444bea83e3bf14278 is the first bad commit
commit 1a2d60afe17bbcc43dfd388444bea83e3bf14278
Merge: 306d493c be5dd5d3
Author: rokath <th@seerose.net>
Date:   Sat Apr 16 17:53:54 2022 +0200

    testing merged into main
```

I'll try to dig deeper into what this problem actually is as time permits, but I'm adding this bug report in case the git bisect output would provide some hints on where to look :)

kraiskil, 2022-05-01:
First part of this failure is the new file `pkg/src/triceConfig.h`, which overshadows the local configuration file of the same name.
Is the `pkg/src/triceConfig.h` intentional, and the `test/OpenCM3_STM32F411_DISCOVERY` demo should use a file of different name? The git log for the new file suggests it was added by mistake :)

Removing `pkg/src/triceConfig.h`and recompiling `test/OpenCM3_STM32F411_DISCOVERY` make the Trice prints show up again, but somehow corrupted. Only the following two lines appear, repeatedly once per second, in the output:

```
May  1 19:58:32.345034  /dev/ttyACM0:                     :  57    676675 WARNING:unknown ID  62089 - ignoring trice [99 1 137 242 42 0 0 0]
May  1 19:58:32.345057  /dev/ttyACM0: Hints:Baudrate? Encoding? Interrupt? Overflow? Parameter count? Password? til.json? Version?
```
I did recompile the host trice binary too, delete til.json, and I did not change the baudrate from the working setup.

rokath, 2022-05-02:
Please use v0.48.0 for now.

rokath, 2022-05-14:
Solved in main. See ReadMe.me inside `./pkg/src/inc`.

closed 2022-05-14 by rokath