rokath/trice #278: Trice User Guide: clarification on chapter "Use It"

escherstair, 2022-05-09:
Chapter [Use It](https://github.com/rokath/trice/blob/master/docs/TriceUserGuide.md#23-use-it) of Trice User Guide says

- Copy 3 files to your embedded project:
    `./pkg/src/trice.h`
    `./pkg/src/trice.c`
    `./test/.../triceConfig.h`

but it seems that this is wrong because `trice.c` includes `trice_test.h` too
https://github.com/rokath/trice/blob/7d9c7761c8e5f86a4c0ffc57391dec6f546318e8/pkg/src/trice.c#L4

Should the whole `./pkg/src/ folder` be copied to my embedded project?

rokath, 2022-05-09:
Thanks for this hint. I will checknthat soon.

rokath, 2022-05-10:
I guess I will remove that dependency and put it into the `triceConfig.h`.

rokath, 2022-05-14:
done in main

closed 2022-05-14 by rokath