rokath/trice #279: Better handling of  ## __VA_ARGS__

escherstair, 2022-05-09:
File `trice.h`
https://github.com/rokath/trice/blob/7d9c7761c8e5f86a4c0ffc57391dec6f546318e8/pkg/src/trice.h#L224-L225
defines `COUNT_ARGUMENTS(...)` macro using `## __VA_ARGS__`
The comment says
`//! The expression `## __VA_ARGS__` ist not supported by older compilers.`
but this is not 100% right.
As far as I know, `## __VA_ARGS__` when `__VA_ARGS__` is empty is handled in a special way by GCC extension (see [here](http://gcc.gnu.org/onlinedocs/gcc-4.5.1/gcc/Variadic-Macros.html#Variadic-Macros)), that strips the comma `,` before `##`.

But this behavior is not standard C, even if it has been proposed for standardization (see [here)](https://www.open-std.org/jtc1/sc22/wg14/www/docs/n2023.pdf).
And so, even a "new" clang compiler (not GCC) doesn't support this GCC extension and it requires a manual change to the code as described by the comment
https://github.com/rokath/trice/blob/7d9c7761c8e5f86a4c0ffc57391dec6f546318e8/pkg/src/trice.h#L224

Please, consider that [Arm Compiler 6](https://developer.arm.com/downloads/-/arm-compiler-for-embedded) is clang-based (not GCCI) and so even the new relase 6.18 (releases on May 2022) requires this change.

I suggest to investigate (and maybe improve) the `COUNT_ARGUMENTS(...)` macro so that it doesn't rely over a GCC extension.
I try to propose a working solution, even if the readability of the code would be a little be worse.

rokath, 2022-05-09:
Thanks a lot for your help. I am not a C macro expert.

escherstair, 2022-05-10:
I have an answer.
Even if it's not part of the C standard, both GCC and clang support the following extended behavior
> if the variable arguments are omitted or empty (and so `__VA_ARGS__` expands to nothing), the `##` operator causes the preprocessor to remove the comma before it.
This it what is used by macro `TRICE(id, fmt, ...)` to work as expected.

The only scenario where this doesn't work is not an old compiler (not realistic today IMHO), but it's when the C language is set to strict-C (i.e. C90, C99, C11 or C17 - strict conformance to standard ISO 9899).
Usually this is done through the command line switch `-std` (`-std=c99`, `-std=c11`, ...).
When the language is set to strict C, a macro named `__STRICT_ANSI__` is defined. This can be used to detect which language is used.
If no `-std` switch is specified, both GCC and clang use the `gnu` language mode (and not strict C).
And so the trice macros work as expected.

For this reason, even if the trice macros are not srict C, they're supported by the majority of the scenarios today.
IMHO changing them in a standard way requires a lot of "black magic macro" that is against the readability of the code.

Probably the best solution is to change a lilttle bit the comment, from
https://github.com/rokath/trice/blob/aa70d63f4daad3ef545934ec2a62d31c1498aeed/pkg/src/trice.h#L224
to
```
//! If you set the C language to strict C (C90, C99, C11 or C17) the `##` operator doesn't remove the comma before it when `__VA_ARGS__` expands to nothing.
//! In this case, the TRICE macro doesn't work with no parameters. You must explicitly use TRICE0 instead of TRICE for a no parameter value TRICE in this case.
```

What do you think?

rokath, 2022-05-14:
Done in main.

closed 2022-05-14 by rokath

jsls135, 2025-08-13:
hi,  rokath & escherstair
First of all, I would like to express my sincere gratitude for your contributions. This is truly an excellent project. 
But regarding this issue (even though it has been closed), I wonder if we can adopt the following solution. The current Trice0 approach breaks the rules of the API and is not convenient for transplantation for an existing project. 
**My idea is:** 
Remove the second parameter fmt of the trice macro and directly merge it into the variable parameter list.

#define TRICE_COUNT_ARGUMENTS(...) TRICE_NTH_ARGUMENT(dummy, ##__VA_ARGS__, 11, 10, 9, 8, 7, 6, 5, 4, 3, 2, 1, 0, dummy)
#define TRICE(  tid, ...) TRICE_CONCAT2(TRICE_,   TRICE_COUNT_ARGUMENTS(__VA_ARGS__))(tid, ##__VA_ARGS__)

Merging "fmt" means that __VA_ARGS__ has at least one parameter, unless the user only calls an empty function (like: printf();), which is not allowed, and the cmd: "trice insert" will do nothing for such empty function call.

This method can process both scenarios (even with strict C).