rokath/trice #282: trice tool doesn't handle '\t'

escherstair, 2022-05-10:
In C source code I use `\t` when I need to send a `tab` character.
As an example
`TRICE("\tVersion %d.%d.%d", 2, 3, 0);`
When I use the provided trice tool (under Windows 10), the `\t` is shown as character `\` followed by character `t`
[image]
and not as a `TAB` (as in a standard RS232 terminal connected to tjhe UART).

Is this expected?
Thanks in advance.

rokath, 2022-05-10:
That is new for me. `\n` works. I need to investigate that. Did you had a look at the til.json for the `\t`? The strings there are the ones **Go** gets for displaying. I guess I forgot to handle `\t` in the  `update` code. Right now I am very short of time but I will solve that.

rokath, 2022-05-10:
Seems not to be a big issue: [https://stackoverflow.com/questions/59952721/how-to-read-t-as-tab-in-golang](https://stackoverflow.com/questions/59952721/how-to-read-t-as-tab-in-golang).

rokath, 2022-05-14:
done in main

closed 2022-05-14 by rokath