rokath/trice #283: TRICE_S macro clarification

escherstair, 2022-05-10:
I need a clarification on the usage of `TRICE_S` macro.
I see [this paragraph](https://github.com/rokath/trice/blob/master/docs/TriceUserGuide.md#dynamic-stringsbuffers-only-as-variable-inside-trice-macros) and as far as I understand, it means "No string literals as arguments inside TRICE macro".
Is this the meaning of the title "Dynamic strings/buffers only as variable inside TRICE macros"? (Sorry, but English is not my native language).
I see that this is a No-Good example
https://github.com/rokath/trice/blame/aa70d63f4daad3ef545934ec2a62d31c1498aeed/docs/TriceUserGuide.md#L833-L836
and a workaround i suggested.

But, isn't the following one the same as No-Good example?
https://github.com/rokath/trice/blame/aa70d63f4daad3ef545934ec2a62d31c1498aeed/docs/TriceUserGuide.md#L169

Moreover, is it possible to trnasform the workaround into this?
```C
void f0( void ){ const char* s = "Hello"; TRICE_S( "msg:%s\n", s ); }
```
Trying to follow the C best practise "const-qualify everything".

rokath, 2022-05-10:
The example with "Paul" is wrong. It needs correction.  Your proposal  concerning const is from the C language point totally correct.  But if the strings known at compile time,  they should get a part of the format strings.

escherstair, 2022-05-11:
I think now I better understand.

All the string literals (i.e. compile-time know strings) should be put inside the format string.
Only the dynamic strings should be used as variables in `TRICE_S` macro.

As an example:
```C
char dynamic_string[50];
fillDynamicStringFromSomewhere(dynamic_string);
TRICE_S( "msg:This part of the string is known at compile time. This part is dynamic %s\n", dynamic_string); 
```

@rokath Can you confirm this is the right usage?
Thanks in advance for all you support.

rokath, 2022-05-11:
That's perfekt! I will take that in the doc.

rokath, 2022-05-14:
done in main

closed 2022-05-14 by rokath