rokath/trice #285: trice tool: send data over UART

escherstair, 2022-05-15:
Sometimes UART is used both to receive data from the microcontroller and to send data to it.
If the microcontroller uses trice to send data, on the PC trice tool is running the handle the incoming packets.
Is there a way to send some data (not trice-encoded,I mean raw data) over the same UART from trice tool.
If trice tool is running, it takes the ownership of the COM port, and Windos doesn't allow me to open another terminal to send data to the microcontroller.
Maybe using display server can help?

rokath, 2022-05-15:
It is not implemented, but easy possible do do by simply writing to the opened comport. There is no protocol defined and the target receive code is not part of trice. Alternatively you could use a comport spiltter on the PC and let your own application write. An other option I see is, to give the trice tool a CLI switch reading from a file and writing every byte to the comport. Then your app could write into this file. The trice tool could get a command window in the future. What exactly is your intension?

escherstair, 2022-05-16:
Thanks @rokath 
My application uses UART to send some diagnostic/debug data (i.e. related to its internal behavior).
And it's possible to send some raw bytes from the PC to the UART to start some specific debug tasks.
Trice is really useful for the received data.
I don't need a complex protocol for transfer (and in any case it is out of trice scope, as ypu wrote), but only sending some bytes.

When you refer to comport splitter, do you think to a Windows application that works as a splitter?

rokath, 2022-05-16:
Never used ia comport splitter, but I saw such stuff somewhere. There is also an open source soluti **com0com** but I think it oly can connect apps over virtual comports. But maybe it can do more.  

For entering commands, **trice** needs a separate additional window. One day it will hopefully have it.

rokath, 2022-05-21:
Mercedes with #291

rokath, 2022-05-23:
A quick idea: `trice ds` in one bash and `trice log -p COM3 -ds` in a 2nd bash. Needs just to implement the write functionality to the target in **trice** and the receiving in the target..

escherstair, 2022-05-24:
> A quick idea: `trice ds` in one bash and `trice log -p COM3 -ds` in a 2nd bash. Needs just to implement the write functionality to the target in **trice** and the receiving in the target..

Do you mean that the write will be done from the DisplayServer?

rokath, 2022-05-24:
The other way around: The DisplayServer will show the logs and the **trice** tool getting the *Trice* messages from the target and transmitting them to the DisplayServer can get your keystrokes and write them down to the target, because it's screen is free. This allows also to select the target to write to by switching the input focus when using several targets. Give me a few days please - too busy with work and family right now.

escherstair, 2022-05-25:
Sure, I don't want to push you in any way.
I can only say thank you for your amazing job, and give my contribution where I'm able to do it

rokath, 2022-06-04:
It should work now with the current main branch and the F030 example.

rokath, 2022-06-06:
Possible now with v0.52.0 release.

closed 2022-06-06 by rokath