rokath/trice #292: Clarification on TRICE_TIMESTAMP

escherstair, 2022-05-27:
In triceConfig.h I see
https://github.com/rokath/trice/blob/8d91c1ee9cbdad0e71f9ba190dbe307ee60410ad/pkg/src/inc/triceConfig.h#L52
but I think that the comment is wrong.
As far as I understand, it should be
either
`//!< Uncomment if you need target timestamps. Instead of SYSTICKVAL, you can use any other up to 32-bit value, like milliSecond.`
or
`//!< Comment if you do not need target timestamps. Instead of SYSTICKVAL, you can use any other up to 32-bit value, like milliSecond.`
as I think it should be enabled if I need timestamps.

Moreover, as far as I understand, `ReadUs32()` should be a hardware dependant implementation of a function that returns the number uf 1us ticks elapsed.
I get this idea looking to some examples
https://github.com/rokath/trice/blob/3d297c1fe471511e2eb7044ed78bddf3beb5a432/test/MDK-ARM_STM32F030R8/Core/Src/main.c#L79-L94
Can you confirm this?

If this is the case, I suggest to change
https://github.com/rokath/trice/blob/8d91c1ee9cbdad0e71f9ba190dbe307ee60410ad/pkg/src/inc/triceConfig.h#L50
into
```C
extern uint32_t ReadUs32( void );
```
and adding a comment for the user.
As a proposal (that can be changed):
```
ReadUs32() must be a function that returns the number uf 1us ticks elapsed.
It's up to the developed providing the proper hardware-dependant implementation of this function.
```

rokath, 2022-05-28:
Your hint concerning the comment it correct. I am going to change the word into 'disable' to avoid irritation. The timestamp timebase is up to the user: clocks or milliseconds or 100 microseconds units or whatever. Future trice will support 32-bit, 16-bit or no timestamp parallel. See https://github.com/rokath/trice/blob/master/docs/TriceV1.0Specification.md#62-main-stream-logs.

rokath, 2022-05-28:
Done in main branch.

closed 2022-05-28 by rokath