rokath/trice #294: In TRICE_MODE == 0 every TRICE macro creates its own stack frame

escherstair, 2022-06-06:
I don't understand if this is related to this issue #203 (since this old issue is marked as completed), but in case `TRICE_MODE == 0` every TRICE macro creates its own stack frame (with variables `co`, `TriceBufferWritePosition` and `tLen`).
This comes from
https://github.com/rokath/trice/blob/4a60e75e144a9f5cabb88b1ffd3080926179f872/pkg/src/inc/triceConfig.h#L76-L85

If this is by design (and cannot be improved in any way), I suggest ot add it into the documentation.

Thank in advance and a big thank you for your great job.

rokath, 2022-06-06:
I cannot see an issue here. Could you please work out a bit more details, what is unclear and/or should improved.

escherstair, 2022-06-07:
Sorry, I didn't explain enough the key point.
I noticed that if I configure
```C
#define TRICE_MODE     0
```
every TRICE macro creates its own variables `co`, `TriceBufferWritePosition` and `tLen` on the stack.
You can see this in the following image from Keil uVision
[image]
For this reason, if you put a lot of TRACE macros inside a function, this requires a big amount of stack (36 bytes for every TRICE macro if `TRICE_STACK_BUFFER_MAX_SIZE` is set to 128).
In this way is quite easy falling into stack overflow (since in embedded devices the stack is not so big).

If this (i.e. every TRICE macro needs to create its own variables on the stack) is mandatory by the design, I suggest adding a warning in the documentation. Something similar to "Do not exceed with the number of TRICE macros inside a function or you will fall into stack overflow".

I don't know if there is a multithread-safe way to improve this. I think that global variables `co`, `TriceBufferWritePosition` and `tLen` used by every TRICE macro is not a multithread-safe approach (in case a high-priority threads preempts a low priority one while this is inside a TRICE macro).

In my first message I refer to #203 because in the descriptio from @th-sabik I read

> Each TRICE macro creates its own stack frame

and so I thought i twas this same problem. The issue is marked as Done, and so probably I'm wrong.

Moreover, I know from the documentation that immediate mode (TRICE_MODE 0) is only for kickstarting with the project and for easily exploring, but it has some limitations.
So. not sure how is important investigating this kind of problems in immediate mode.

escherstair, 2022-06-09:
@rokath I found the reason for the behavior.
I've been using Arm Compiler 6 (clang-based) with optimization set to `-O0`
From ARM documentation, in this case

> Code size and stack usage are also significantly higher
And

> -O1 — restricted optimization. This optimization level provides a good debug experience with better code quality than -O0. Also the stack usage is improved over -O0. Arm recommends this option for a good debug experience.

With this compiler is necessary to use -O1 (or higher), otherwise the stack is not reused and the assuption "the stack frame of the first TRICE is already removed" is false.
Probably this is true for every arm-clang complier, as you can see looking to the assembly [here](https://godbolt.org/z/Webxe4E95) and looking what happens changing optimization level to `-O1`

If this is true, we can add ad advice to Trice users to set optimization level to `-O1` for clang-based compilers (even if this should be done independently by the usage of Trice or not).

rokath, 2022-06-09:
Maybe -O0 is meaningful for compiler debugging. The optimization dependend stack frame construction indeed needs clarification inside the *Trice* documentation. Congratulation for your deep analyzis.

rokath, 2022-06-14:
Documented in [TriceUserGuide.md](https://github.com/rokath/trice/blob/master/docs/TriceUserGuide.md).

closed 2022-06-14 by rokath