rokath/trice #299: Trice not working correctly on BE32 devices

hippiefm, 2022-07-13:
Trice seems not to work correctly on BE32 devices despite setting **TRICE_BIG_ENDIANNESS** flag in `triceConfig.h`

**Expected output**

Having `TRICE( Id(34644),"Test");` inside my code should produce **Test** in console, instead I am getting `ERROR:package len 4 is < 340  - ignoring package [135 84 0 192]`. 
I figured out this needs to be some problem with byte ordering, so I swapped endianness inside `TriceOut` function:
```
...
#ifdef TRICE_ENCRYPT
    eLen = (eLen + 4) & ~7; // only multiple of 8 encryptable
    TriceEncrypt( da, eLen>>2 );
#endif
#ifdef TRICE_BIG_ENDIANNESS
    for (uint32_t i = 0; i < eLen / 4; i++) {
        da[i] = ((da[i] >> 24) & 0xFF) | ((da[i] << 8) & 0xFF0000) | ((da[i] >> 8) & 0xFF00) | ((da[i] << 24) & 0xFF000000);
    }
#endif
cLen = TriceCOBSEncode(co, (uint8_t*)da, eLen);
...
```
Now the output is correct, however from time to time I am getting message like `CYCLE: 196 not equal expected value 198 - adjusting. Now 126 CycleEvent` - the numbers inside are changing. 
Is byte swapping after COBS encoding needed as well?

rokath, 2022-07-14:
The cycle error usually occurs, when messages are lost, what can happen if you produce more messages than the hardware can deliver. You could add your own cycle counter as parameter to check that. When BE32 is a big endian device you are the first user. COBS encoding is eniannes agnostic. It only handles single bytes. You could also disable the cycle counter for testing.

rokath, 2022-07-14:
Please change `elen / 4` into `(elen >> 2)`. I do not know if your compiler is clever and divisions are expensive.

rokath, 2022-07-14:
I am not sure about your change proposal. Do you use the CLI switch `-targetEndianess`?

hippiefm, 2022-07-14:
You are absolutely right, I wasn't aware of `-targetEndianess` switch existence - switching it to `bigEndian` works like a charm. Thank you very much for you help.

closed 2022-07-14 by hippiefm