rokath/trice #325: No ID generation for multi-line TRICE statements

moud93, 2022-08-29:
**Description**
If a TRICE statement takes more than a line, `trice u` doesn't generate any IDs for it

**To Reproduce**
Steps to reproduce the behavior:

1. Add the following code example:
```
1  TRICE("This is a one line statement\n");
2  TRICE("This is a two line" 
3           "statement\n");
```
2. Use the command `trice u` to add IDs
3. Verify the output:
- Statement at line 1 will be replaced with:
```
1  TRICE( Id(1234),"This is a one line statement\n");
```
- Statement at line 2 will stay the same  
```
2  TRICE("This is a two line" 
3           "statement\n");
```

**Expected behavior**
An ID for statement at line 2 must be generated
```
2  TRICE( Id(1235), "This is a two line" 
3           "statement\n");
```

**Desktop**

- OS: Fedora Linux 35 (Workstation Edition)
- Linux Kernel: 5.18.16-100.fc35.x86_64
- Version: v0.18.4-1136-gbfc1e39f

rokath, 2022-08-29:
This is a known limitation. See here: https://github.com/rokath/trice/blob/master/docs/TriceUserGuide.md#105-limitation-trice-u-requires-trice-macros-on-a-single-line
Anyway, thanks for opening this issue. It needs to be fixed one day, but has currenly low priority.

rokath, 2023-07-24:
Done with `trice insert` in v0.61.0

closed 2023-07-24 by rokath