rokath/trice #326: Compiler adaption improvements in trice_config.h

escherstair, 2022-09-12:
trice_config.h has a section Compiler Adaption
https://github.com/rokath/trice/blob/b739c672f31361ad0199e136a812d765f2bcf557/pkg/src/inc/triceConfig.h#L163-L213

It seems that it differentiates between gnu compiler and ARMkeil.
The problem is that ARM Keil can use two different compilers, really different :

- ARM Compiler 5 (AC5)
- ARM Compiler 6 (AC6)

The predefined macros for the two compilers are different:

- predefined macros for AC5 are listed [here](https://developer.arm.com/documentation/dui0491/i/Compiler-specific-Features/Predefined-macros). `__arm__` is always defined. `__GNUC__` is defined only if you specify `--gnu` command line option (to compile in gnu mode)
- predefined macros for AC6 are listed [here](https://developer.arm.com/documentation/100067/0612/Other-Compiler-specific-Features/Predefined-macros) . `__arm__` is defined when targeting AArch32 state with `--target=arm-arm-none-eabi`. On the other side, `__GNUC__` is alway set.

Moreover `cmsis_armcc.h` exists for AC5 only. AC6 should include `cmsis_armclang.h`

I confirm that with the actual `defines` my code, compiled with AC6 falls into
https://github.com/rokath/trice/blob/b739c672f31361ad0199e136a812d765f2bcf557/pkg/src/inc/triceConfig.h#L167-L169
and this is not what expected (based on what I see in the following lines).

I think that AC6 should be handled in a different way, as an example based on `__ARMCC_VERSION` macro.

I'm going to propose a pull request

th-sabik, 2022-09-15:
Can we close this issue now?

escherstair, 2022-09-16:
yes

closed 2022-09-16 by rokath