rokath/trice #361: Can it support zephyrrtos？

hongshui3000, 2023-01-18:
Now zephyr supports external custom logs, it will be very interesting if trice can be used as an external library on zephyr

rokath, 2023-01-20:
Never worked with zephyr. Please have a look at [zephyr dictionary-based-logging](https://docs.zephyrproject.org/3.1.0/services/logging/index.html#dictionary-based-logging) as an option. If you intend to use trice as external library you could ask the zephyr experts. Compiling trice as a library would speed up the build process, I guess, but take code size optimization options away. This is with the master branch no problem, because it only uses code inline macros. The `noids` branch (not usable yet!) includes the option to use function calling macros and these functions would be all compiled into a library, even you do not need them. Maybe a kind of library config could be a solution then. Hope that helps.

hongshui3000, 2023-01-21:
thanks，I see

closed 2023-01-21 by hongshui3000