rokath/trice #365: No trice binary file

Andy79881, 2023-01-22:
After cloning I could not find the trice binary or do I have to build it from sources? As mentioned the build from sources is optional and this step can be skipped.

Andy79881, 2023-01-22:
Building trice from source worked for me. Maybe the documentation shall be updated accordingly. Furthermore for TDM-GCC installation a hint would be helpful which version is required. There are basically two versions avaliable:

- 64+32-bit MinGW-w64 edition
- 32-bit-only MinGW.org edition

I have downloaded and installed the first one. But maybe the 32-bit only version is sufficient.

rokath, 2023-01-22:
Thanks Andy for your interest!
- Did you read [this](https://github.com/rokath/trice/blob/master/docs/TriceUserGuide.md#21-get-it)?
  - What about [https://github.com/rokath/trice/releases](https://github.com/rokath/trice/releases)? The binaries are usually not part of the sources.
- Concerning GCC: [https://github.com/rokath/trice/blob/master/docs/TriceUserGuide.md#3-build-trice-tool-from-go-sources-you-can-skip-that](https://github.com/rokath/trice/blob/master/docs/TriceUserGuide.md#3-build-trice-tool-from-go-sources-you-can-skip-that) Is this sufficient? I did not mention 32- or 64- bit version, because I assumed that it logically should match the installed Go variant. But according to your hint, I added this now.

rokath, 2023-01-24:
When installing from sources, the trice binary is in folder user/go/bin. After downloadinģ, you can place it anywhere. On Windows you could create C:\bin and extend your path variable accordingly. In Linux /home/username/bin usually exists.

Andy79881, 2023-02-04:
issue is solved from my point of view. Thanks

closed 2023-02-04 by rokath