rokath/trice #366: Linker error

Andy79881, 2023-01-22:
I get the following linker errors:

-------------- Build: Debug in L051_trice_test ---------------

Compiling: src\startup_stm32l051xx.S
Compiling: src\system_stm32l0xx.c
Compiling: src\timing.c
Compiling: src\trice.c
Compiling: src\main.c
Linking executable: bin\Debug\L051_trice_test.elf
c:/program files (x86)/embitz/2.50/share/armgcc/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: obj\debug\src\trice.o: in function `TriceOut':
C:\Users\USER\Documents\embitz_projects\STM32L051_Trice_test/src/trice.c:221: **undefined reference to `triceBuffer'**
c:/program files (x86)/embitz/2.50/share/armgcc/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Andreas\Documents\embitz_projects\STM32L051_Trice_test/src/trice.c:221: **undefined reference to `triceBufferDepth'**
c:/program files (x86)/embitz/2.50/share/armgcc/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: obj\debug\src\main.o: in function `main':
C:\Users\USER\Documents\embitz_projects\STM32L051_Trice_test/src/main.c:162: **undefined reference to `ReadUs16'**
collect2.exe: error: ld returned 1 exit status
Process terminated with status 1 (0 minutes, 3 seconds)
3 errors, 0 warnings (0 minutes, 3 seconds)

The trice.h is included into my file, trice update works well. I do not know what could be the error.

rokath, 2023-01-23:
`ReadUs32` and `ReadUs16` the user has to provide. They are project specific. Check the F030 test project for examples.  This should translate error free and give you a hint concerning the missing `triceBufferDepth.` What are your settings in `triceConfig.h`?

rokath, 2023-02-04:
No further information.

Andy79881, 2023-02-04:
issue is still not solved for me. Structure of triceConfig.h not completely understood. I do not use any LL or HAL driver. I have a project using bare metal approach (so CMSIS used only). If you can please provide a working trice.Config.h for F030 this would help.

rokath, 2023-02-04:
Currently a new release gets prepared in the branch noids. There the F030 project translates and in the main branch it should translate as well. I recommend to clone the repo. Install uVision from ARM. It is free usable, with STM M0 cores but needs an additional free license if image size is > 32kb. You could also remove the triceCheck.c file to make the image smaller. When using an UART, you need to provide the access functions. These are now with the LL drivers but you can have your own implementation. In direct mode (stack buffer) you only need a putchar(). Or, ging the RTT way, you do not need UART access at all.

rokath, 2023-02-04:
The uVision is proposed just for checking and getting started. Then switch to your preferred tool chain.

rokath, 2023-02-04:
Please use the triceConfig.h from the F030 project and tell me what you do not understand so that I can improve the comments.  There are many changes between main and noids branch.

rokath, 2023-02-04:
I would like to improve the doc/TriceUserGuide.md file but for that I neede to understand what exacty you are missing.

breakersun, 2023-02-10:
I have the same problem too. I also compared the triceConfig.h from F030 with \pkg\src\inc\triceConfig.h, not find any defines for `ReadUs32` or `ReadUs16`. 

> `ReadUs32` and `ReadUs16` the user has to provide. They are project specific. Check the F030 test project for examples. This should translate error free and give you a hint concerning the missing `triceBufferDepth.` What are your settings in `triceConfig.h`?

rokath, 2023-03-04:
Can we close this?

Andy79881, 2023-03-05:
Unfortunately I had no time to test. There seems to be a solution as 'breakersun' mentioned. So, please, yes you can close this issue.

closed 2023-03-05 by Andy79881