rokath/trice #382: failed to start over RTT channel

space-blaster, 2023-02-16:
**Describe the bug**
got runtime error while starting trice.exe over uart.

**Screenshots**
[image]

Error messages:
```
panic: runtime error: slice bounds out of range [:6] with capacity 0

goroutine 1 [running]:
github.com/rokath/trice/internal/trexDecoder.(*trexDec).Read(0xc0001ac000, {0xc0001e6000, 0x10000, 0x10000})
        C:/repos/trice/internal/trexDecoder/trexDecoder.go:249 +0xef2
github.com/rokath/trice/internal/translator.decodeAndComposeLoop({0xb25440, 0xc00008a2b8}, 0xc0000d0150, {0xb25e88, 0xc0001ac000}, 0xa556e0?, 0x7cab35?)
        C:/repos/trice/internal/translator/translator.go:108 +0xc8
github.com/rokath/trice/internal/translator.Translate({0xb25440?, 0xc00008a2b8}, 0xa9cf00?, 0xc000089200, 0xc00009b1e8, 0xc000089380, {0xb26350?, 0xc0000e6360})
        C:/repos/trice/internal/translator/translator.go:79 +0xa75
github.com/rokath/trice/internal/args.logLoop({0xb25440?, 0xc00008a2b8})
        C:/repos/trice/internal/args/handler.go:184 +0x7fe
github.com/rokath/trice/internal/args.Handler({0xc00010e000, 0x5, 0x8})
        C:/repos/trice/internal/args/handler.go:97 +0x254
main.doit({0xb255a0, 0xc00000a018})
        C:/repos/trice/cmd/trice/main.go:41 +0x14c
main.main()
        C:/repos/trice/cmd/trice/main.go:29 +0x27
```

rokath, 2023-02-16:
Quick Answer now (I will later check more details):

Deactivate the trice messages after the HEADLINE. Create an own trice message and run `trice u` before compiling. 
You can also set the Id(12345) values to Id(0). I guess it is due to a not matching til.json file.

Further hints/questions: 
* What is your hardware?
* You talk about UART but use JLINK?
* You have STMF031K6 in your command line but I see STMF030R8 in the output?
* When using JLINK, the trice mode 0 is recommended.
* You could try the `noids` branch - that´s where I am currently working on, to be hopefully soon released.

rokath, 2023-02-17:
After checking out `master` and changing `triceConfig` I could run on a **NUCLEO-STMF03R8**:
[image]
It is interesting to know, how your setup differs.

rokath, 2023-02-17:
Please, which `trice.exe` you are using? Run `trice version` for that. I cannot locate a slice at `trexDecoder.go:249`.

rokath, 2023-02-17:
The panic is possible, when more `%d` in the format string than parameters are given. You could do:
```bash
git switch noids
go install ./...
git switch -
```
Than you can work normally with `master` but use the newer `trice.exe` which should give you a hint, if my assumption is right.

breakersun, 2023-02-19:
> Please, which `trice.exe` you are using? Run `trice version` for that. I cannot locate a slice at `trexDecoder.go:249`.

`version=0.55.1, commit=bfc1e39fdb323d5fff8077782083c7ca6f9ef635, built at 2022-08-25T20:02:36Z`

Will try noids branch a little bit later.

rokath, 2023-02-19:
The panic is caused due to invalid input data. This is the relevant code line in trexDecoder.go:249
```Go
n += copy(b[n:], fmt.Sprintln("WARNING:unknown ID ", triceID, "- ignoring trice", p.B[:p.TriceSize]))
```
`p.TriceSize` is here 6 but `p.B` has already length 0 and that causes the panic.

The `noids` branch avoids any assumptions with its error message in trexDecoder.go:313:

```Go
n += copy(b[n:], fmt.Sprintln("WARNING:unknown ID ", triceID, "- ignoring trice ending with", p.B))
```

As already said, please check your `TRICE` messages and your `til.json` file. The command `trice u` with the right switches **before** compiling is mantadory. Then when calling `trice l` make sure you use the same `til.json` file.

breakersun, 2023-02-20:
Thank you @rokath for explaining this.
In my case, if I give any ID bigger than 20000(roughly), it will cause problem like I posted.
If I keep the trace ID smaller than 10000(roughly), it works fine.
I doubt there is such limitation, didn't figure out why.

Also, `trice.exe u` is executed before compiling each time, so there is nothing wrong with til.json or li.json.

breakersun, 2023-02-20:
> Thank you @rokath for explaining this. In my case, if I give any ID bigger than 20000(roughly), it will cause problem like I posted. If I keep the trace ID smaller than 10000(roughly), it works [...]

Sorry this is a false alarm, somehow `trace u` doesn't update IDs in C code in early test. Now IDs are updated in C, and logs output are good.

closed 2023-02-20 by space-blaster

rokath, 2023-02-20:
That's fine. Thaks for the feedback. I opened a new issue #386.

rokath, 2023-02-20:
Last night branch `noids` was Mercedes into `master`.