rokath/trice #387: UART Stoped Output on STM32F030C8T6

breakersun, 2023-02-21:
Version:
trice.exe version=0.56.0, commit=0a94b2c7a7172c6a06a37e7b14ed96c3bc545d1c, built at 2023-02-20T10:00:58Z
other .c/.h from main repo, commit id: 7a0710739f42564c441973dd959caac8763c8723

Problem:
1. prepare dependency .c/.h, comment out TRICE_RTT0 in triceConfig.h
2. trice.exe u
3. build and flash target board
4. trice.exe l -p COM6 -baud 921600
5. output a few lines then stopped
[image]

Tried:
1. debug with jlink, looks it always stopped at this line after booting up a few seconds
triceModbusBuffer.c
```
static uint32_t* triceNextModbusBuffer( size_t len ){
    if( TriceBufferWritePosition > triceModbusBufferWriteLimit ){
        for(;;); // trice modbus buffer overflow // here it stopped and never get out
    }
    if( (int)triceModbusBufferWriteLimit - (int)TriceBufferWritePosition > len ){
        return triceModbusBufferWritePosition; // enough space at buffer end
    }else{
        return triceModbusBufferHeap; // buffer wrap
    }
}

```

rokath, 2023-02-21:
You can comment out MODBUS in triceConfig.h line 118. The host side is not implemented yet. Despite of this you see the right output. If you like, you can include triceCheck.c and change the `#if 0` to `#if 1` in main.c.

rokath, 2023-02-21:
I will check later more details

rokath, 2023-02-21:
Yes,
```C
    if( TriceBufferWritePosition > triceModbusBufferWriteLimit ){
        for(;;); // trice modbus buffer overflow // here it stopped and never get out
    }
```
is wrong. It is now commented out and also the target modbus code is deactivated per derfault in Release v0.56.3

rokath, 2023-02-21:
Please close this issue, if it works now.

breakersun, 2023-02-21:
thank you, now it works.

closed 2023-02-21 by breakersun