rokath/trice #390: Define TRICE_OFF in a single point

escherstair, 2023-02-22:
[Here](https://github.com/rokath/trice/blob/master/docs/TriceUserGuide.md#10-target-side-trice-on-off) I read that adding

> you can add #define TRICE_OFF just before the #include "trice.h" line and no trice code is generated anymore for that file

But this means that I have to add `#define TRICE_OFF` to every file where I included `trice.h`
Would be possible having a way to define `TRICE_OFF` for the whole project?
As an example, passing `TRICE_OFF` definition to the preprocesso with an option?
GCC uses `-D` as described [here](https://gcc.gnu.org/onlinedocs/gcc/Preprocessor-Options.html) but every toolchain has its own way.

rokath, 2023-02-22:
Good point, but `-D TRICE _OFF` seems to be a general possible way, as far as I know.

escherstair, 2023-02-24:
Based on my preliminary test, it seems that it works

rokath, 2023-02-25:
It is now in the [user manual](https://github.com/rokath/trice/blob/master/docs/TriceUserGuide.md#10-target-side-trice-on-off). Thanks for the hint.

rokath, 2023-03-04:
Can we close this?

escherstair, 2023-03-04:
yes

closed 2023-03-05 by rokath