rokath/trice #402: Trice l -p FILE..., works but not, trice l -p TCP4...

jersuNF, 2023-03-30:
**Describe the bug**
Using the same setup, workspace and configuration I see some errors when connecting trice to the JLinkExe but not when using trice to read from a file created by JLinkLogger.

**Creating file and using trice to read it, works Good!**
<terminal_2>$: JLinkRTTLogger -Device NRF52840_XXAA -if SWD -Speed 4000 -RTTChannel 0 triceRaw.log
<terminal_2>$: trice l -p FILE -args triceRaw.log
Mar 30 11:08:02.548133  FILE:                main.c    62              Starting Nofence application...
Mar 30 11:08:02.548183  FILE:                main.c    95              another message 

**Creating JLinkExe session and connecting trice by TCP4, works Bad!**
In the example below I open the JLinkExe in another terminal and keep it open, once trice is connected I have to issue "reset" followed by "go" into the "J-Link>" program.

<terminal_1>$: cat  ~/script_base/bash/collar_utils/connect_nrf52840.jlink
device NRF52840_XXAA
si SWD
speed 4000
<terminal_1>$: JLinkExe -autoconnect 1 -CommandFile ~/script_base/bash/collar_utils/connect_nrf52840.jlink
J-Link>

<terminal_2>$:  trice l -p TCP4 -args localhost:19021
inconsistent TCOBSv1 buffer: [83 69 71 71 69 82 32 74 45 76 105 110 107 32 86 55 46 56 50 100 32 45 32 82 101 97 108 32 116 105 109 101 32 116 101 114 109 105 110 97 108 32 111 117 116 112 117 116 13 10 83 69 71 71 69 82 32 74 45 76 105 110 107 32 65 82 77 32 86 49 49 46 48 44 32 83 78 61 56 50 49 48 49 49 48 53 55 13 10 80 114 111 99 101 115 115 58 32 74 76 105 110 107 69 120 101 13 10 18 82 192 35 0]
Mar 30 11:23:49.862105  TCP4:                main.c    95              CYCLE: 193 not equal expected value 192 - adjusting. Now 1 CycleEvents
Mar 30 11:23:49.862203  TCP4: another message 

**Versions of code and tools**
commit 9977bb5cbc3a1f5b129cb2a8c0d5fa270e4e6fea (HEAD -> master, origin/master, "renamed back accidently rename"

go version go1.20.2 linux/amd64

Ubuntu 20.04

**Expected behavior**
I expect to see the same output from trice regardless if it is reading from a file or connected to JLinkExe (also known as JLink commander)

**Screenshots**
[image]

**Additional context**
I have tried this from bash terminal, spacemacs vterm, and inside VScode, all the same results.

rokath, 2023-03-30:
JLinkRTTLogger seams to be the only usable tool for trice. The others expect ASCII strings and use 0xFF n as a terminal selector if I remember right. They filter the 0xFF n out and destroy the data stream. I am unhappy with this situation but found no other way. By the way ay: when using `trice l -p JLINK -args "-device ..."` a temporary file inside temp is generated automatically and trice uses it to read from. When applying the `-v` switch you see the details.

rokath, 2023-03-30:
Have a look into https://github.com/rokath/trice/blob/master/docs/TriceOverRTT.md for more information, if not already done.

rokath, 2023-03-30:
Is here more to do? Can we close this?

rokath, 2023-04-05:
Seems to be the same as in https://github.com/rokath/trice/issues/403#issuecomment-1496136098

closed 2023-04-11 by jersuNF