rokath/trice #407: Disabling color still prints color in the summary

krigney-exyn, 2023-04-20:
First, thanks for the excellent tool! We've been very happy with how it's been working for us.

**Describe the bug**
When you disable color with the log tool, using `-color off` or `-color none` you still receive colorized output after you ctrl-c.

You can also see a bug in this screenshot that there is no space between the timestamp and "default:"

Here's a redacted screenshot:
[image]

Thanks!

rokath, 2023-04-20:
Thanks for the bug report! To switch color off at the end I simply forgot. As workaround until it is fixed I recommend `trice ... | grep -a -v " times: "`, in the hope the string " times: " is not part of your logs.

With `-color none` the channel info should disappear. Not so with `-color off`. 
No space before "default:" is intended:
[image]
If there would be a space, the black space between blue and green would get blue too.
So, can you live with `-color none`?
[image]

krigney-exyn, 2023-04-21:
The lack of a space between the timestamp and `default:` was inconsistent with the other fields (`info:`, `time:`) so I figured it was a bug.

As for the channel info with `-color off`, I wasn't expecting to see `info:`, `time:`, or `default:` on every log line, only the channel info for the message text. I understand that switching `-color none` discards _all_ the channel info. What I didn't expect was seeing the extra channel info for the line number and timestamp. I didn't expect those fields (line number, timestamp) to have channel info associated with them. It would be nice if I could see the channel info for *only* the message text, not the other metadata in the message.

rokath, 2023-04-21:
Am I understand you right, that you wish removed channel info generally until the trice string starts, when you use `-color off` and `-color none`? That is possible of course. Please let me understand your use case a bit better. 

If you wish to tag your trice messages right now, 
* you could use any "green_tag:" or "red_tag:" instead of "debug:" for example. These not matching tags are not touched with `-color none`. 
* you could use at least one upper case letter like "Debug:" or "DEBUG:" with `-color none`
* you could edit https://github.com/rokath/trice/blob/master/internal/emitter/lineTransformerANSI.go - all used channels only and only in this file - and build the trice tool.

Please give me a bit of time to think about that. I will not break anybodys scripts. Probaly I will add an additional option `-color no` or so which implements your proposal.

rokath, 2023-07-29:
fixed in v0.61.1

commit 6cfcd801 (2023-07-29): fixed #407
closed 2023-07-29 by rokath