rokath/trice #412: Missing data using RTT with JLinkRTTLogger and Trice

gjfernandezcaballero, 2023-05-10:
Hello. I'm using Trice as logger into a RTOS system project. I have a task with a counter that only prints the value and increments the variable, into a loop of 10ms aprox. You can see it in the image below. That's only with testing purposes. The program is running in debug mode using _**SEGGER Ozone**_ and, to read the log messages, I use Trice command in my PC by execute the following commands in two different terminals, as in [Issue #403](https://github.com/rokath/trice/issues/403). 

[image]

Terminal 1: `/usr/bin/JLinkRTTLogger -Device MIMXRT1176XXXA_M7 -if JTAG -Speed 4000 -RTTChannel 0 triceRaw.log`
Terminal 2: `trice l -v -pf none -p FILE -args triceRaw.log`

The output of these terminals are as follow:

[image]
[image]

The issue is that, as you can see, in the output of the second terminal some messages are missed. I've checked the triceRaw.log file and some message are missed too. Also, I've checked in the RTT buffer into RTOS memory and all the data is present. It seems that JLinkRTTLogger reads corrupted messages and writes them to the file, so Trice can't decode them well. 

[image]

I would like to know if you or someone else has the same problem or knows how to solve it.
Thank you.

rokath, 2023-05-10:
Right away no idea. Here things seem to work with Windows:
[image]
I know that JLinkRTTViewer "steals" some bytes assuming these are control sequences: https://github.com/rokath/trice/blob/master/docs/TriceOverRTT.md#44-jlinkrttviewerexe. But that is not the case here because JLinkRTTLogger is used.

You could try to use `TRice` instead of `trice` and set inside your `triceConfig.h` the TriceStamp32 to increment a counter. What is strange: Usually the TRICE_CYCLE_COUNTER is enabled and if a message is missing the trice tool complains about it. Is it possible that your counter gets incremented somwhere else too?

To get started maybe you use `-packageFraming TCOBS` first with the appropriate settings inside `triceConfig.h`.

Also you could  use the `-s` and `-debug` switches to see the incoming bytes.

I would not recommned to mix ASCII strings with trice data, so the `printf` statement should not be there.

gjfernandezcaballero, 2023-05-11:
First of all, than you for your fast response.

I have tried all the things you told me, but the result is the same than before. Now, I can see the lost messages in Trice, but the problem continues. I have updated the JLink software package to the last version (v7.88b), and same. You can see it in the attached screenshots. 

  
[image]
[image]
[image]

I have been checking the output of the Segger device via its interface in localhots:19080/rtt.htm and all the messages can be found. In the following screenshot you can see as, in the left the 0x10 message exists but in the right is not received (it must be between 0x0f and 0x11).

[image]

rokath, 2023-05-11:
Do you have the chance to try it out temporary on a Windows PC, so that we can find out if it is OS related?

gjfernandezcaballero, 2023-05-12:
Hello, I finally was able to read all the messages without missing anyone. Simply, I tested it again using my own PC, with Ubuntu installed too, and everything works fine. The problem was that I was doing the tests using a virtual machine in another computer with more than one virtual machine running at the same time. The SEGGER is connected to this computer, and set to only one host, but maybe the multiplexation of the port could be causing this issue, I'm not sure. Whatever, it's working very fine!

[image]

Has you can see in the picture above, for 38385 messages, no one was missing. 

Now, I have another topic to talk about, but I'm going to open another issue for that. 
Thank you very much for your time and dedication. 
Regards.