rokath/trice #416: Configure Trice to send messages via UART

gjfernandezcaballero, 2023-05-12:
Hello, I'm software developer and I'm using Trice logger in a RTOS system project.

Right now, I'm using Trice over Segger RTT and it works very fine, but I would like to use Trice over UART too. I have been checking the triceConfig.h file and reeding all the instructions, but no one works properly for me. I obtain compilation errors and I can't see nothing via UART in my PC. I have been searching also for documentation about this topic and I didn't find anything. 

The main idea of my project is to use Trice logger over Segger RTT, but just in case that this method won't be available, use Trice over UART port. 

Could you please help me with that question?
Thank you.

rokath, 2023-05-12:
./test/MDK-ARM_STM32F030R8_instrumented/Core/Inc/triceConfig.h inside the latest release is probably what you want. It transfers directly via RTT32 without package framing and parallel via UARTA with package framing TCOBS. You can receive both trich streams parallel using 2 trice tool instances in different terminals.

gjfernandezcaballero, 2023-05-13:
I have tried with your triceConfig file recomendation, but compilation process fails. It gives me two types of compiling errors that you can see below

[image]
[image]

The problem is that configuration variables are not compatible. Specifically the options in the screenshot below

[image]

In your triceConfig.h file, UARTs are not define, and it produces a "error wrong configuration", so it doesn't work. 

On the other hand, I'm using the [v0.58.0](https://github.com/rokath/trice/releases/tag/v0.58.0) version of Trice, and this works perfect via RTT, as you know from the other issue. But I've tried to download the last version in master branch and, despite it compile perfectly, I cannot see nothing via RTT nor UART.

rokath, 2023-05-14:
This is actually work in progress. 0.58 is not the latest release. Please download 0.59 and use exacltly the `triceConfig.h` I mentioned in https://github.com/rokath/trice/issues/416#issuecomment-1546099992. 

Some of your comiler messages are not trice related.

rokath, 2023-05-14:
Just merged the first working twin test into master: https://github.com/rokath/trice/tree/master/test/doubleBuffer_twin_direct_noRouting_nopf_deferred_multi_cobs

Inside automatic tests RTT and UART not usable, so the trice lib writes into a buffer in direct transfer, that then is passed to the trice tool and gets translated there and the test compares the result. In a 2nd step the buffer gets cleared and with "TriceTransfer" the deferred transfer is executed with the same following steps.

So taking the https://github.com/rokath/trice/issues/416#issuecomment-1546099992 `triceConfig.h` and comparing it with the`triceConfig.h` in the twin test should help you to setup your project.

rokath, 2023-05-15:
Please look at https://github.com/rokath/trice/blob/master/CHANGELOG.md#v0601-twice-log-screen-shot.
If everything works well on your mashine pleas close this issuse.

closed 2023-05-25 by rokath