rokath/trice #419: Log messages as string of characters

gjfernandezcaballero, 2023-05-15:
Hello again!

I have another question for you. As you probably know, I'm developer in a RTOS system project, and I'm using Trice as logger. I'm working on a logger library that internally uses Trice or printf, depending on what the user configures. That function of the library receives a char* argument, and I have to use this in TRice function to print it out via RTT.  I've tried two different implementations, but any of them works properly.

First case:
[image]
[image]

Second case:
[image]
[image]

I get a failure in both cases, and I don't understand why. Do you know what is happening?
Thank you!

rokath, 2023-05-15:
First case:

When the trice tool parses the code it assumes a special syntax: `trice( "format string" [,data])` and it must be in one line. You cannot replace "format string" with a variable name.  This is a limitation. The parser is just a simple one and has no capabilities like a preprocessor. The  "format string" goes into the til.json file which is read by the trice tool then.  I know this could be done better and maybe will happen one day.

2nd case:

Strings need special handling. Use `TRICE_S( ID(0), "%s", output_message)`. That option exists for strings not known at compile time, for example when a command is received and echoed. It is not recommended for static strings because then there is no real advantage anymore compared to printf.

rokath, 2023-05-15:
What if you generally wrìte `triCE( "format string")` and in your `triceConfig.h` `#define triCE printf` or `#define triCE trice`? This is just a very idea and needs more elaboration. For that I need a better use case understanding.
There is already a `TRICE_OFF` option. The same way a `TRICE_PRINTF` could be written what looks more promizing.

rokath, 2023-05-16:
When you compile `til.json.c` reduced to the actual used IDs into the target, you could code `TriceNonBlockingDeferredWriteAuxiliary` (see ./src/triceAuxiliary.c)  and execute it as deferred output option performing sprintf in a buffer which you can pass to `triceNonBlockingWriteUartB` for example. Then you can let the user configure that, if needed.

This way it is possible to have direct RTT transfer for developing and parallel deferred printf output just for a specific ID range. Then in the field no trice tool is needed. Any terminal will do. Of course this increases the target code amount.

closed 2023-07-29 by rokath