rokath/trice #432: Trice basic example with Zephyr RTOS

anuraj-rp, 2023-09-19:
**Describe the bug**
I am trying to run the basic example running with trice prints every second with basic instrumented code as below - 
trice( iD(2893), "hello trice \a\n"); 
but I get the following error 
```
error: macro "trice32_1" requires 3 arguments, but only 2 given
trice( iD(2893), "hello trice \a\n");
```

I adapted one of the triceConfig.h from the samples which is attached. I am not able to figure out why it is expanding to 3 arguments. 

[triceConfig.h.txt](https://github.com/rokath/trice/files/12663916/triceConfig.h.txt)

anuraj-rp, 2023-09-19:
Okay, I got it compiling by defining the macro TRICE_CLEAN

`#define TRICE_CLEAN`

Is there any documentation what this macro does?

rokath, 2023-09-20:
```bash
$ trice h -c
syntax: 'trice sub-command' [params]
sub-command 'c|clean': Set all [id|Id|ID](n) inside source tree dir to [id|Id|ID](0). Remove all iD(n).
#       All in source code found IDs are added to til.json if not already there. Inside ti.json differently used IDs are
#       reported and just cleaned inside the source files. The existing li.json is not used. A new li.json is generated in place.
#       The switch "-src" is optional (default is "./") and a multi-flag here. So you can use the "-src" flag several times.
#       Example: 'trice clean -src ../A -src B/x.c': Sets all TRICE IDs to 0, or removes them, in folder ../A. and file B/x.c
#       EXPERIMENTAL! The command itself works reliable, but a sophisticated editor will detect inconsistencies with removed IDs,
#       EXPERIMENTAL! if macro TRICE_CLEAN is not defined before "#include "trice.h". For that a good idea is needed.
```
Focus: *a sophisticated editor will detect inconsistencies with removed IDs*. So you should not use `TRICE_CLEAN` - it is only a first idea to silence editor warnings automatically. In fact when using it and you compile, the trice macros are getting empty and your error seem to be gone. `TRICE_OFF` exists to disable all trice messages in a file, what is pretty the same as `TRICE_CLEAN`. There is a very high chance, that the macro `TRICE_CLEAN` will disapear sooner or later.

- As a quick workaround write `trice0( "Hi" );` if you hve no values as parameters.

I assume the problem is this one: lets look a bit deeper into **trice.h** lines 683-691 and read the comment:

```C
//! COUNT_ARGUMENTS builds upon NTH_ARGUMENT. The more arguments that are passed to COUNT_ARGUMENTS, 
//! the more the »counting arguments« (12, 11, 10, 9, 8, 7…) are pushed to the right. 
//! Thus the macro evaluates to the number of arguments that are passed to the macro.
//! If you set the C language to strict C (C90, C99, C11 or C17) the `##` operator doesn't remove the comma before it when `__VA_ARGS__` expands to nothing.
//! In this case, the TRICE macro doesn't work with no parameters. You must then explicitly use TRICE0 instead of TRICE for a no parameter value TRICE.
//! For more details see closed Issue #279. Special thanks @escherstair.
//! If for example using CLANG 6.18 set C-language to gnu11, gnu99 or std to avoid the comma issue when no parameters are in a TRICE  macro.
//! In case you have to set the C-Language to c11 or c99 you can use the TRICE0 macro directly instead of TRICE when no value parameters.
#define COUNT_ARGUMENTS(...) NTH_ARGUMENT(dummy, ## __VA_ARGS__, 12, 11, 10, 9, 8, 7, 6, 5, 4, 3, 2, 1, 0)
```

So changing the compiler C-language can solve your problem. If you cannot change the compiler C-language for some rason, you need to use the workaround.

rokath, 2023-09-23:
Can we close this?

anuraj-rp, 2023-09-24:
Thanks for the clarification regarding TRICE_CLEAN macro. 

Yes this issue can be closed.

closed 2023-09-25 by th-sabik

KammutierSpule, 2024-03-30:
If someone is coming on this post because Zephyr, check:
https://devzone.nordicsemi.com/f/nordic-q-a/94829/how-can-i-use-a-newer-c-standard-c11-or-c17-when-compiling-my-zephyr-application

Consider add this
set_property(GLOBAL PROPERTY CSTD gnu17)