rokath/trice #446: Switch from UART Deferred to RTT Direct

initdb, 2024-02-12:
**Describe the bug**
I migrated our trice logging from uart deferred write to rtt direct write. the trice tool cant parse some messages and discards them.

**To Reproduce**
Steps to reproduce the behavior:

1. triceConfig:
```c
//! TriceStamp16 returns a 16-bit value to stamp `Id` TRICE macros. Usually it is a timestamp, but could also be a destination address or a counter for example.
//! The user has to provide this function. Defining a macro here, instead if providing `int16_t TriceStamp16( void );` has significant speed impact.
#define TriceStamp16() timedate_get16bittime()

//! TriceStamp32 returns a 32-bit value to stamp `ID` TRICE macros. Usually it is a timestamp, but could also be a destination address or a counter for example.
//! The user has to provide this function. Defining a macro here, instead if providing `int32_t TriceStamp32( void );` has significant speed impact.
#define TriceStamp32() timedate_get32bittime()

//!  TRICE_BUFFER selects, where the TRICE macros accumulate the trice data during a single TRICE execution. Selectable options:
//! - TRICE_STACK_BUFFER: No additional buffer is needed, what makes sense for single task systems with direct output only.
//! - TRICE_STATIC_BUFFER: A single trice is stored in a separate static buffer, what makes sense in single- and multi-tasking systems with direct output only.
//! - TRICE_DOUBLE_BUFFER: TRICE macros write direct into a double buffer without any additional management action.
//!   This is the fastest execution option for TRICE macros but needs more RAM. Used for deferred output and optional additional direct output.
//! - TRICE_RING_BUFFER: TRICE macros write direct into a ring buffer without any additional management action.
//!   This is not the fastest execution option for TRICE macros but needs less RAM. Used for deferred output and optional additional direct output.
//! If unsure select TRICE_DOUBLE_BUFFER. The TRICE_RING_BUFFER option works, but is experimental.
#define TRICE_BUFFER TRICE_STATIC_BUFFER

//! TRICE_DIRECT_OUTPUT == 0: only deferred output, usually UART output only
//! TRICE_DIRECT_OUTPUT == 1: with direct output, usually RTT
//! Setting TRICE_BUFFER to TRICE_STACK_BUFFER or TRICE_STATIC_BUFFER demands TRICE_DIRECT_OUTPUT == 1, no deferred output at all.
//! When TRICE_BUFFER == TRICE_RING_BUFFER or TRICE_BUFFER == TRICE_DOUBLE_BUFFER for deferred output, additional direct output can be switched on here.
//! For example it is possible to have direct 32-bit wise RTT TRICE_FRAMING_NONE output and deferred UART TRICE_FRAMING_COBS output.
//! TRICE_BUFFER == TRICE_STACK_BUFFER or TRICE_BUFFER == TRICE_STATIC_BUFFER needs TRICE_DIRECT_OUTPUT == 1.
#define TRICE_DIRECT_OUTPUT 1

//! TRICE_DATA_OFFSET is the space in front of single trice data for in-buffer (T)COBS encoding.
//! - When using real big buffers, 16 may be not enough.
//! - When having only short trices but lots of trice bursts, it may make sense to reduce this value to 4.
//! - Without encoding/framing this value can be 0.
#define TRICE_DATA_OFFSET 0 // must be a multiple of 4

//! TRICE_SINGLE_MAX_SIZE is used to truncate long dynamically generated strings and to detect the need of a ring buffer wrap.
//! - Be careful with this value: When using 12 64-bit values with a 32-bit stamp the trice size is 2(id) + 4(stamp) + 2(count) + 12*8(values) = 104 bytes.
//! - In direct mode, and also when you enabled TRICE_SEGGER_RTT_8BIT_DEFERRED_WRITE, this plus TRICE_DATA_OFFSET is the max allocation size on the target stack with TRICE_BUFFER == TRICE_STACK_BUFFER.
//! - When short of RAM and, for example, max 2 32-bit values with a 32-bit stamp are used, the max trice size is 2 + 4 + 2 + 2*4 = 16 bytes.
//! - You should then also disable all then forbidden trices to avoid mistakes. Example: `#define ENABLE_TRice32fn_3 0` and so on at the end of this file.
#define TRICE_SINGLE_MAX_SIZE (1*512) // must be a multiple of 4

//! TRICE_DEFERRED_BUFFER_SIZE needs to be capable to hold trice bursts until they are transmitted.
//! When TRICE_BUFFER == TRICE_STACK_BUFFER this value is not used.
//! When TRICE_BUFFER == TRICE_STATIC_BUFFER this value is not used.
//! When TRICE_BUFFER == TRICE_DOUBLE_BUFFER, this is the sum of both half buffers.
//! When TRICE_BUFFER == TRICE_RING_BUFFER, this is the whole buffer.
#define TRICE_DEFERRED_BUFFER_SIZE (2*(512+32)) // must be a multiple of 4

//! TRICE_MCU_IS_BIG_ENDIAN needs to be defined for TRICE64 macros on big endian MCUs for correct 64-bit values and 32-bit timestamp encoding-
//#define TRICE_MCU_IS_BIG_ENDIAN

//! TRICE_DIRECT_OUT_FRAMING defines the framing method of the binary trice data stream for direct output. Options:
//! - TRICE_FRAMING_TCOBS: Recommended for internal transfer and trice tool visualization.
//! - TRICE_FRAMING_COBS: The trice tool needs switch `-pf COBS`. Useful with XTEA or to decode the binary trice data with a user tool.
//! - TRICE_FRAMING_NONE: The trice tool needs switch `-pf none`. TRICE_FRAMING_NONE is needed for fast RTT (32-bit access), recommended.
//! - With TRICE_SEGGER_RTT_32BIT_DIRECT_WRITE == 1 or TRICE_SEGGER_RTT_8BIT_WRITE_DIRECT_WITHOUT_FRAMING == 1,
//!   the RTT data arrive unframed ignoring the TRICE_DIRECT_OUT_FRAMING setting here.
#define TRICE_DIRECT_OUT_FRAMING TRICE_FRAMING_NONE

//! TRICE_DEFERRED_OUT_FRAMING defines the framing method of the binary trice data stream for deferred output. Options:
//! - TRICE_FRAMING_TCOBS: Recommended for UART transfer and trice tool visualization.
//! - TRICE_FRAMING_COBS: The trice tool needs switch `-pf COBS`. Useful with XTEA or to decode the binary trice date with Python or an other language.
//! - TRICE_FRAMING_NONE: The trice tool needs switch `-pf none`. This mode may be helpful if you write your own trice viewer without a decoder.
#define TRICE_DEFERRED_OUT_FRAMING TRICE_FRAMING_NONE
[... 60 lines cut ...]
 157 0 0 61 0 174 157 174 157 0 0 62 16 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 50 149 50 149 0 0 63 0 180 197 0 0 0 0 64 5 68 69 66 85 71 0 0 0 255 203 0 0 0 0 65 9 68 78 83 95 83 84 65 67 75 0 0 0 39 214 0 0 0 0 66 13 115 97 118 101 95 100 101 102 97 117 108 116 115 0 0 0 165 220 0 0 0 0 67 4 100 110 115 51 112 157 112 157 0 0 68 0 174 157 174 157 0 0 69 16 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 50 149 50 149 0 0 70 0]                                                                               
Feb 12 18:39:03.590431  JLINK: Hints:Baudrate? Encoding? Interrupt? Overflow? Parameter count? Format specifier? Password? til.json? Version?
```
7. See error

**Expected behavior**
this is the output i receive over uart (just the logfile):
```
Nov 29 15:07:10.478738  COM4: [0;1;36;109m        app_threadx.c   129 [0m[0;7;34;104m   0,000_000[0m [0m==================================================[0m
Nov 29 15:07:10.479287  COM4: [0;1;36;109m        app_threadx.c   130 [0m[0;7;34;104m       0_000[0m [0mhardware revision:	B2[0m
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

- OS: win 11 pro
- JLink SW + RTT sources version 7.88c
- trice version v62.1, v62.3 and master branch show the same result

**Additional context**
I tried different revisions and settings. i suspected a race condition but the locking functions are present and working. the problem appeard with TRice32 and TRICE_S macros. the calls with no param seem to work more likely. also some with param work...at the moment any idea would be welcome.

Kind regards

rokath, 2024-02-12:
Hi @initdb did you consider the `-d16` switch?

```bash
trice h -l
...
  -d16
        Short for '-Doubled16BitID'.
 ...
  -doubled16BitID
        Tells, that 16-bit IDs are doubled. That switch is needed when un-routed direct output is used like (TRICE_SEGGER_RTT_32BIT_DIRECT_WRITE == 1), but also with double buffer in (TRICE_TRANSFER_MODE==TRICE_PACK_MULTI_MODE) and XTEA encryption. Read the user guide for more details.
```
If that is not the point, here are some general hints:
- Your triceConfig.h looks good but we are searching an error:
  - Please set TRICE_DATA_OFFSET to 16 just for curiosity. 0 should be fine.
  - Please set TRICE_DEFERRED_BUFFER_SIZE to 0. Any other value should be OK.
  - `#define TRICE_SEGGER_RTT_8BIT_DIRECT_WRITE 1` and undefine TRICE_SEGGER_RTT_32BIT_DIRECT_WRITE. The 32-bit version I wrote by myself and if Segger changes s.th. internal, the 8-bit version should still work.
  - Some general hints brcause I have no better ideas right now:
    -  Use trice macros only in a single task and do that slowly. If things start to work, increase speed and use multi tasking.
    - Copy and paste trice lines from triceCheck.c to your code and try separately 0-bit, 16-bit and 32-bit timestamps,

rokath, 2024-02-12:
https://github.com/rokath/trice/blob/master/docs/TriceUserGuide.md#152-package-format contains now  `-d16` switch information and also the triceConfig.h files got an additional comment line. (Right now only in main branch).

rokath, 2024-02-13:
BTW: There is no need to "switch". You can have UART deferred and RTT direct both together.

initdb, 2024-02-13:
the `-d16` switch improved it a little bit, i will try the different configs. we want to only use RTT, since our UART was to slow/ we log a lot.

initdb, 2024-02-13:
the proposed config changes further improved the parsing. Now i suspect that it just might be the speed, which is causing the issues. I will try to increase the rtt buffer. is there any other way to increase the output speed? i saw the dubble buffer which might be faster than the static buffer?

rokath, 2024-02-13:
> the `-d16` switch improved it a little bit, i will try the different configs. we want to only use RTT, since our UART was to slow/ we log a lot.

RTT is best during development, but it is normally not usable in the field.

rokath, 2024-02-13:
> the proposed config changes further improved the parsing. Now i suspect that it just might be the speed, which is causing the issues. I will try to increase the rtt buffer. is there any other way to increase the output speed? i saw the dubble buffer which might be faster than the static buffer?

A double buffering is internally a bit faster than a ring buffer for the price of memory but that counts only for **deferred modes**. The internal speed is always much higher than any interface could handle.

I do not know exactly the max RTT speed, but I have the impression, that a professional J-Link works better than a converted ST-Link (https://github.com/rokath/trice/blob/master/docs/TriceOverRTT.md#41-convert-a-stm-nucleo-or-discovery-onboard-st-link-valid-for-st-link-v2--v21-not-for-v3). Also I have seen cases, where the converted ST-Link internally came out of order and after a power cycle was everything fine again. That happened especially under high trice loads. So reducing the data amount for investigation seems to be a good strategy. Use *trice8* or *trice16* when possible to reduce data amount.

Increasing the RTT buffer inside SEGGER_RTT_Conf.h is a real good idea. This avoids buffer overruns when bursts occur. Also use `TRICE_SEGGER_RTT_8BIT_DIRECT_WRITE` until you got it stable.

initdb, 2024-02-13:
we plan to make the jtag interface switchable(not permanently disabled) and try to get away with only this interface, we could switch back to uart if the need arrives.

initdb, 2024-02-13:
we only use jlink classic plus, i will experiment a bit and get back with the results.

initdb, 2024-02-13:
it is stable now with the `TRICE_SEGGER_RTT_8BIT_DIRECT_WRITE 1` or the `TRICE_SEGGER_RTT_32BIT_DIRECT_WRITE 1` flag. the RTT buffer size `BUFFER_SIZE_UP` is now 4096.  Thank you❤️ 

the parallel deferred output via uart would be nice, if there is an option to just output to a certain log level.

i also observed some hangups on sleep of our controller, if no debugging session was previously active or just at random. I will investigate.

rokath, 2024-02-13:
You can define `TRICE_UARTA_MIN_ID` and `TRICE_UARTA_MAX_ID` inside triceConfig.h to log only a specific ID range over UARTA in deferred mode. The trice tool is planned to be extended for channel specific ID ranges. Right now you need to do the ID assignment partially manually.

initdb, 2024-02-15:
ok, thank you very much. We will keep this in mind once we get further. At the moment im developing a skript to watch the output of trice with the JLinkRTTLogger and restart in the device goes to sleep.

rokath, 2024-02-15:
If you do s.th. universal useable - pull requests are welcome.

rokath, 2024-02-15:
Can we close this issue?

initdb, 2024-02-17:
we still have an inconsistency on different machines, we will send it to you once its working.
yes we can close the issue, thank you very much!

rokath, 2024-02-18:
Assuming the *"inconsistency on different machines"* is s.th. different, this issue gets closed now. Feel free to re-open it or to create a new issue anytime. The SEGGER stuff is good, but should not be considered as error free per default. You could contact them for support, if applicable, and also compare alternatives [openOCD](https://wiki.segger.com/OpenOCD), [Bluepill](https://stm32-base.org/boards/STM32F103C8T6-Blue-Pill.html).

closed 2024-02-18 by rokath