rokath/trice #448: v0.62.3 build broken with RESTRICT change?

jrperrin, 2024-03-04:
**Describe the bug**
For release v0.62.3 I get a compile error in cobsDecode.c due to 'RESTRICT' not being valid as a compiler directive. RSTRICT is interpreted as the name of the void* parameter and the compiler errors out on the unexpected 'out'.

**To Reproduce**
Try to compile release v0.62.3 with a compiler that does not use RESTRICT like arm-none-eabi-gcc (as opposed to __restrict).

**Expected behavior**
I expect trice to build and define RESTRICT, it appears cobs.h needs `#include "triceConfig.h"`

**Additional context**
The master branch appears to have reverted some of the restrict changes and also included triceCofig.h in cobsDecode.c but not in cobs.h. 

Am I doing something wrong or does v0.62.3 have a bug? I think I could fix this locally by including triceConfig.h in cobs.h but I don't want to deviate from the repo or fork it if I don't have to.

Thanks,
Joe

rokath, 2024-03-05:
Thanks for reporting this issue, Joe! 

This is indeed a problem not easy to handle in a clean way for  any compiler. I think, **cobs.h** or **tcobs.h** should not do `#include triceConfig.h`. Therefore they contain now `__restrict` directly. My assumption is, that, if the user's compiler does not know the `__restrict` keyword (pre C99), he is able to define that in the project settings. Right now I commented out all `RESTRICT` occurrences in the main branch, to remove them later. The `RESTRICT` removal was work in progress I forgot to finish.

The remove of `#include <stdint.h>` from **trice.h** has its root cause, that it should be part of the **triceConfig.h** file. Unfortunately I forgot to do that in most examples. This is now fixed in the main branch.

I hope to make a small release within the next days after testing and some feedback from you.

rokath, 2024-03-05:
It is difficult to help with just the logs. Please provide your triceConfig.h file together with the command line and a few logs. You can additionally use the `-s` switch. 
Some general hints:
- Could it be, that you configured COBS in the first example? The trice tool assumes TCOBS per default, If you configured COBS, you have to tell that the trice tool via CLI `-pf COBS` as you did with `-pf none` in the 2nd example.
- Are you sure your bytes arrive correct? (baud rate, parity, stop bit(s)) Default is115200 8N1. I see  {115200 8 0 0 <nil>}. Could that mean no stop bit or what is the second 0 for?
- When switching package framing off, you might need to use the `-d16` switch as well. Switching package framing off makes sense usually when using RTT in direct mode. I did not test intensively other cases and cannot guaranty they all work fine.

rokath, 2024-03-05:
You could also write in your code `trice( iD(170), "hello\n");`. Then you should see `aa` as one byte in the data stream.

rokath, 2024-03-05:
"trice in sync with master or one of the releases?" - for production code stay with releases. Use master only if needed.

closed 2024-03-05 by jrperrin

rokath, 2024-03-05:
Happy to see, all is ok now. May I ask what went wrong before ?

jrperrin, 2024-06-27:
Sorry, I didn't mean to leave you hanging. Everything has been working very well with commit 4ae88445. I'm a little fuzzy on the details now but it looks like we did the followinng:
1. Switched from using `TRICE_SEGGER_RTT_32BIT_DIRECT_WRITE` to `TRICE_SEGGER_RTT_8BIT_DEFERRED_WRITE` instead. 
2. Added a `#define RESTRICT` to our triceConfig.h. 
3. Used `SEGGER_RTT_LOCK()` for `TRICE_ENTER_CRITICAL_SECTION`
 
We can then launch trice with `trice l -p TCP4 -args localhost:9090` 

I've pasted the whole triceConfig.h below. Thank you very much for this project.

``` c
// clang-format off
/*! \file triceConfig.h
\author Thomas.Hoehenleitner [at] seerose.net
*******************************************************************************/

#ifndef TRICE_CONFIG_H_
#define TRICE_CONFIG_H_

#ifdef __cplusplus
extern "C" {
#endif

#include <stdint.h>

// HACK(lawrenr): This variable is defined in FreeRTOSv10_0_0/Source/tasks.c. It keeps track of the number of
//                FreeRTOS ticks that have occurred. FreeRTOS considers this to be privileged data, but I
//                but I want to use it to keep track of the number of milliseconds elapsed. This should NOT
//                be used outside of FreeRTOS and this file.
extern volatile uint32_t xTickCount;

// HACK(lawrenr): These give the value of the systick reload and systick current value registers. These could
//                (and probably should) also be referenced by the CMSIS interface, but I didn't feel like
//                adding them to the trice_config target includes. Fix this if you want.
#define systick_reload (*(volatile uint32_t *) 0xE000E014UL)
#define systick_value  (*(volatile uint32_t *) 0xE000E018UL)

#define RESTRICT //!< Adapt it to your compiler: restrict, __restrict, __restrict__ or empty

//! TriceStamp16 returns a 16-bit value to stamp `Id` TRICE macros. Usually it is a timestamp, but could also be a destination address or a counter for example.
//! The user has to provide this function. Defining a macro here, instead if providing `int16_t TriceStamp16( void );` has significant speed impact.
#define TriceStamp16() (0x1616)

//! TriceStamp32 returns a 32-bit value to stamp `ID` TRICE macros. Usually it is a timestamp, but could also be a destination address or a counter for example.
//! The user has to provide this function. Defining a macro here, instead if providing `int32_t TriceStamp32( void );` has significant speed impact.
// NOTE(lawrenr): FreeRTOS uses Cortex M4 systicks to maintain its system tick. SysTick is a countdown timer that updates every clock cycle, or once every 1/48th
//                of a microsecond. FreeRTOS configures this register to reset every millisecond. With this in mind, we can take the difference between the systick
//                reload and systick value registers to get the number of cycles since the last reload. We can then divide this by 48 to get microseconds since
//                last reload. Finally, we can use xTickCount to measure milliseconds.
#define TriceStamp32() (xTickCount * 1000l + (systick_reload - systick_value) / 48l)

//!  TRICE_BUFFER selects, where the TRICE macros accumulate the trice data during a single TRICE execution. Selectable options:
//! - TRICE_STACK_BUFFER: No additional buffer is needed, what makes sense for single task systems with direct output only.
//! - TRICE_STATIC_BUFFER: A single trice is stored in a separate static buffer, what makes sense in single- and multi-tasking systems with direct output only.
//! - TRICE_DOUBLE_BUFFER: TRICE macros write direct into a double buffer without any additional management action.
//!   This is the fastest execution option for TRICE macros but needs more RAM. Used for deferred output and optional additional direct output.
//! - TRICE_RING_BUFFER: TRICE macros write direct into a ring buffer without any additional management action.
//!   This is not the fastest execution option for TRICE macros but needs less RAM. Used for deferred output and optional additional direct output.
//! If unsure select TRICE_DOUBLE_BUFFER. The TRICE_RING_BUFFER option works, but is experimental.
#define TRICE_BUFFER TRICE_DOUBLE_BUFFER

//! TRICE_DIRECT_OUTPUT == 0: only deferred output, usually UART output only
//! TRICE_DIRECT_OUTPUT == 1: with direct output, usually RTT
//! Setting TRICE_BUFFER to TRICE_STACK_BUFFER or TRICE_STATIC_BUFFER demands TRICE_DIRECT_OUTPUT == 1, no deferred output at all.
//! When TRICE_BUFFER == TRICE_RING_BUFFER or TRICE_BUFFER == TRICE_DOUBLE_BUFFER for deferred output, additional direct output can be switched on here.
//! For example it is possible to have direct 32-bit wise RTT TRICE_FRAMING_NONE output and deferred UART TRICE_FRAMING_COBS output.
//! TRICE_BUFFER == TRICE_STACK_BUFFER or TRICE_BUFFER TRICE_STATIC_BUFFER needs TRICE_DIRECT_OUTPUT == 1.
#define TRICE_DIRECT_OUTPUT 0

//! TRICE_DATA_OFFSET is the space in front of single trice data for in-buffer (T)COBS encoding.
//! - When using real big buffers, 16 may be not enough.
[... 354 lines cut ...]
#define ENABLE_TRice64fn_3  1
#define ENABLE_TRice64fn_4  1
#define ENABLE_TRice64fn_5  1
#define ENABLE_TRice64fn_6  1
#define ENABLE_TRice64fn_7  1
#define ENABLE_TRice64fn_8  1
#define ENABLE_TRice64fn_9  1
#define ENABLE_TRice64fn_10 1
#define ENABLE_TRice64fn_11 1
#define ENABLE_TRice64fn_12 1

#endif // #if TRICE_32_BIT_SUPPORT

//
///////////////////////////////////////////////////////////////////////////////

#ifdef __cplusplus
}
#endif

#endif /* TRICE_CONFIG_H_ */

// clang-format on

```