rokath/trice #449: Trice (or JLink RTT) exits with status 255

jolyons123, 2024-03-06:
**Describe the bug**
The exact same setup (microcontroller, source code) works on my colleagues Windows machines but does not work on my Ubuntu setup.

In particular, I have JLink SEGGER tools version 7.94l and newest trice version. Whenever I try to run the `./tools/linux/trice l -p JLINK -pf none -d16 -args '-Device STM32L552ZE -if SWD -Speed 4000 -RTTChannel 0'` command, it looks like "something" exits with 255 just about one second after executing the above command. Executing the `JLinkRTTLogger -Device STM32L552ZE -if SWD -Speed 4000 -RTTChannel 0` command gives no errors and runs just fine.

**To Reproduce**
Steps to reproduce the behavior:

1. Have li.json and til.json in current working directory
2. Run the `trice l -p JLINK -pf none -d16 -args '-Device STM32L552ZE -if SWD -Speed 4000 -RTTChannel 0'` command

**Expected behavior**
The trice program should print the log messages.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

- OS: Ubuntu 22.04.4 LTS

rokath, 2024-03-06:
Please add `-debug` to the CLI and send the logs. It could be the background process does not start correctly. Maybe a path issue.

jolyons123, 2024-03-07:
It seems like the invocation of the JLinkRTTLogger executable within the trice process is not able to find the RTT Control Block!

See the log:

```./tools/linux/trice l -debug -v -p JLINK -pf none -d16 -args '-Device STM32L552ZE -if SWD -Speed 4000 -RTTChannel 0'

No logfile writing...
Read ID List file til.json with 328 items.
Read ID location information file li.json with 2577 items.
port: JLINK arguments: -Device STM32L552ZE -if SWD -Speed 4000 -RTTChannel 0
LINK executable JLinkRTTLogger and dynamic lib JLinkARM expected to be in path for usage.
Start a process: JLinkRTTLogger with needed lib JLinkARM and args:
0 -Device
1 STM32L552ZE
2 -if
3 SWD
4 -Speed
5 4000
6 -RTTChannel
7 0
8 temp/trice-2811516626.bin
/usr/bin/JLinkRTTLogger -Device STM32L552ZE -if SWD -Speed 4000 -RTTChannel 0 temp/trice-2811516626.bin
SEGGER J-Link RTT Logger
Compiled Feb 29 2024 10:30:03
(c) 2016-2017 SEGGER Microcontroller GmbH, www.segger.com
         Solutions for real time microcontroller applications

Default logfile path: /home/john/.config/SEGGER

------------------------------------------------------------ 

------------------------------------------------------------ 
Connected to:
  SEGGER J-Link
  S/N: 602001526

Searching for RTT Control Block...**RTT Control Block not found**. Cannot get data.

Shutting down... Done.trice is watching and reading from temp/trice-2811516626.bin
Encoding is TREX
exit status 255```

whereas the direct invocation of the JLinkRTTLogger correctly finds the RTT Control block (both invocations use the same args):

```JLinkRTTLogger -Device STM32L552ZE -if SWD -Speed 4000 -RTTChannel 0                                                
SEGGER J-Link RTT Logger
Compiled Feb 29 2024 10:30:03
(c) 2016-2017 SEGGER Microcontroller GmbH, www.segger.com
         Solutions for real time microcontroller applications

Default logfile path: /home/john/.config/SEGGER

------------------------------------------------------------ 

------------------------------------------------------------ 
Connected to:
  SEGGER J-Link
  S/N: 602001526

Searching for RTT Control Block...OK.
3 up-channels found:
0: Terminal
1: 
2: 
Selected RTT Channel description: 
  Index: 0
  Name:  Terminal
  Size:  4096 bytes.

Output file: /home/john/.config/SEGGERRTTLogger_Channel_Terminal.log

Getting RTT data from target. Press any key to quit.
------------------------------------------------------------ 

Transfer rate: 208 Bytes/s Data written: 2.37 KB```

rokath, 2024-03-07:
PLease check https://github.com/rokath/trice/issues/403.

jolyons123, 2024-03-07:
> PLease check #403. Could this be the same issue?

Yes this is the same issue. I tried the JLinkGDBServer approach but this did not work for me. Instead, the [workaround](https://github.com/rokath/trice/issues/403#issuecomment-1491661669) you showed works just fine! Thanks for your help!

closed 2024-03-07 by jolyons123