rokath/trice #452: error: macro "TRICE32_1" requires 3 arguments, but only 2 given

KammutierSpule, 2024-03-30:
**Describe the bug**
ct2401.cpp:51:8: error: macro "TRICE32_1" requires 3 arguments, but only 2 given
   51 |         TRice( iD(5720), "😴\n" );

**Additional context**

compiling with Zephyr, c++ file.

rokath, 2024-03-31:
Please read https://github.com/rokath/trice/issues/279. I guess you need to change your compiler settings or to use `TRice0` as workaround.

KammutierSpule, 2024-04-01:
Yes, that is related, and this
https://github.com/rokath/trice/issues/432#issuecomment-2028209322
helped me to fix on a C file.

However, I noticed that the problem seems to be related with other included files.

CONCAT and other defines are messing with the build.

rokath, 2024-04-01:
The information you provided is not sufficient for me for an action. How to deal further with this issue?

closed 2024-04-01 by KammutierSpule