rokath/trice #456: bug / feature request: add TRICE_ prefix to ID's

KammutierSpule, 2024-04-05:
I'm getting the following issues while using trice and C++:

```
[build] In file included from /usr/include/c++/11/bits/ios_base.h:41,
[build]                  from /usr/include/c++/11/streambuf:41,
[build]                  from /usr/include/c++/11/bits/streambuf_iterator.h:35,
[build]                  from /usr/include/c++/11/iterator:66,
[build] /usr/include/c++/11/bits/locale_classes.h:516:10: error: expected unqualified-id before ‘{’ token
[build]   516 |     id() { }
```

rokath, 2024-04-05:
Would you please explain the problem and proposal in more detail?

In general,  as pointed out in https://github.com/rokath/trice/issues/439, trice macros inside header files are not recommended.

KammutierSpule, 2024-04-06:
I'm using Trice on a C++ project.
Some of the source code is header only based, so I need to include trice on the .h file.

I'm experience some of defines used in Trice are in conflict with other defines on other libraries: iD, ID, CONCAT,
So it would be nice if you could add a TRICE_ prefix to
#define TRICE_iD
#define TRICE_ID
#define TRICE_CONCAT
...

rokath, 2024-04-06:
`TRICE_CONCAT` will be no problem. To `TRICE_`-prefix `ID`, `Id`, `id` and `iD` could be not a so good idea. This would worsen the user code readability. With which library are the name clashes? Is that adaptable? Using `#include "trice.h"` inside other header files is not recommended. Could it be possible to use `#undef ID` before using `#include "trice.h"` in a .CPP file?

KammutierSpule, 2024-04-06:
The conflict with iD is when other (eg: c++ std library files) are included after "trice.h".
So on this case I can't undef iD as it will be need for trice.

Sometime is possible to rearrange the includes and workarround it, but gets harder to fix (or impossible?) when there are multiple include files that need trice.h

```
fileA.hpp
#include "otherstuff.hpp"
#undef CONCAT
#include "trice.h" <-- ok as trice is included later
```

```
fileB.hpp
#include "fileA.hpp"
#include "anotherstuff.hpp" <-- conflict here with "iD" because trice was already include on fileA.hpp
#include "trice.h" 
```

Possible workaround:
```
fileB.hpp
#include "anotherstuff.hpp" <-- include before fileA
#include "fileA.hpp"
#include "trice.h" 
```

Possible workaround:
```
fileB.hpp
#include "fileA.hpp"
#undef iD
#include "anotherstuff.hpp"
#include "trice.h"  <-- include trice.h so it will define iD again
```

rokath, 2024-04-07:
fileA.hpp:
```C
#undef iD
#include "otherstuff.hpp"
#undef iD
#include "trice.h" 
```
fileB.hpp:
```C
#include "fileA.hpp"
#undef iD
#include "anotherstuff.hpp" 
#undef iD
#include "trice.h" 
```
Could that help as workaround?

rokath, 2024-04-12:
In master branch now some renamings to avoid name clashes. Only ID, Id, id, iD stay unchanged for user code compability and readability. They are undefined now at the trice.h beginning.

closed 2024-04-12 by rokath