rokath/trice #464: check for invalid use of Trice0

KammutierSpule, 2024-05-16:
Trice0( variable ); // invalid use
Trice0( "tex %u", variable );  // invalid use

Should the trice tool flag this invalid usage of the Trice0 ?

rokath, 2024-05-16:
Usually you do not need to put the parameters count into the trice name. `Trice0(` is as good as `Trice(` when compiling with at least c99. See also issue #279. The code snippets you showed do not contain `iD(123)`. I would expect the compiler to complain about that. If you run `trice i`, the trice tool should insert the IDs. Then you'll compile and if you dislike the IDs in the code, you can remove them with `trice c` afterwards until next compilation.

KammutierSpule, 2024-05-16:
The use of Trice0 is related to https://github.com/rokath/trice/issues/452 issue (compile on C++)

```
Trice0( variable ); // invalid use
Trice0( "tex %u", variable ); // invalid use
```

On these cases, trice insert is not giving an ID or report any message. it should report this mistake.

rokath, 2024-05-16:
Such trice tool check is a good idea, but it would need an additional parsing which would slow down the whole process. Also it is no solution for the limits of the current parsing because some other weird sequence could occur as well. I assume, that the compiler anyway will complain about missing IDs and the developers will be able to fix the code.

KammutierSpule, 2024-05-16:
Actually what was happening to me was something like:
Initial state:
```
Trice0( variable ); // invalid use
Trice0 ( "Good Trice\r\n" );
```
trice insert
```
Trice0( id(id), variable ); // No compiler complain ( don't mind id vs ID vs Id vs iD)
Trice0 (  id(id), "Good Trice\r\n" );
```
trice clean
```
Trice0( variable );
Trice0 (  id(id), "Good Trice\r\n" ); // it keeps the previous if, no clean
```
I'm not sure if this was the sequence but somehow it ended with "orphan" ID's.
I detected that because I was getting unknown Ids in the code (after I manually clean the json files)

rokath, 2024-05-17:
When you clean the til.json file you need execute `trice i` afterwards, before you compile.

KammutierSpule, 2024-05-17:
I just tested this, started from a empty li.json/til.json

```
void trice0_test() {
    Trice0( "OK");
    Trice( InvalidUse );
    Trice( "OK", Variable );
}
```

```
void trice0_test() {
    Trice0( iD(2740), "OK"); // ok, iD is added
    Trice( InvalidUse ); // no warning or error
    Trice( "OK", Variable ); // id is not added / updated
}
```

rokath, 2024-05-17:
Yes, that is  consistent wrong.

rokath, 2024-05-19:
In https://github.com/rokath/trice/blob/master/docs/TriceUserGuide.md#922-linited-trice-parser-capabilities there is now this documented as limitation. I close this issue now as we still have #465 and a separate check makes not much sense as explained.

closed 2024-05-19 by rokath

rokath, 2024-08-17:
> The use of Trice0 is related to #452 issue (compile on C++)
> 
> ```
> Trice0( variable ); // invalid use
> Trice0( "tex %u", variable ); // invalid use
> ```
> 
> On these cases, trice insert is not giving an ID or report any message. it should report this mistake.

The 2nd example would be covered on solving issue #150.