rokath/trice #465: invalid use of Trice0 will misbehave trice insert

KammutierSpule, 2024-05-16:
Trice0( variable ); // invalid use
Trice0( "text %u", variable ); // invalid use

Will cause that the next Trices will be ignored or not updated.

See also: https://github.com/rokath/trice/issues/464

rokath, 2024-05-16:
Am I understand you right, that the first line causes the trice tool not to insert further IDs in that file? That's indeed an issue. Unfortunately I am not a parser expert and the trice tool is somehow limited.

There is probably no quick fix right now. See also comment in *https://github.com/rokath/trice/blob/master/internal/id/insertIDs.go* near line 280.

KammutierSpule, 2024-05-16:
also, it happens with clean too.
If there is an invalid use of Trice0, trice clean will not be able to clean next ids on the file

rokath, 2024-05-19:
It is added no into https://github.com/rokath/trice/blob/master/docs/TriceUserGuide.md#922-linited-trice-parser-capabilities.

rokath, 2024-11-13:
The following sequence inside `trice/examples/OpenCM3_STM32F411_Nucleo/main.c`
```C
               // Depending on mode, either print this string to
		// UART (mode 0), or the Trice write buffer (mode 200).
		TRICE( Id( 4030),"Hello, TRICE, %d\n", 42);

		// TRICE() with a string parameter only is problematic.
		// See discussion on https://github.com/rokath/trice/issues/279
		// TRICE0() works in either case
		#ifdef __STRICT_ANSI__
		// if compiled with e.g. --std=c99
		TRICE0( Id( 4683),"Hello, TRICE\n");
		#else
		TRICE( Id( 2926),"Hello, TRICE\n");
		TRICE0( Id( 7263),"Hello, TRICE0()\n");
		#endif
```
gets changed into
```C
		// Depending on mode, either print this string to
		// UART (mode 0), or the Trice write buffer (mode 200).
		TRICE(Id(0), "Hello, TRICE, %d\n", 42);

		// TRICE(Id(0), "Hello, TRICE\n");
		#else
		TRICE(Id(0), "Hello, TRICE\n");
		TRICE0(Id(0), "Hello, TRICE0()\n");
		#endif
```
so these lines are disappearing:
```C
		// TRICE() with a string parameter only is problematic.
		// See discussion on https://github.com/rokath/trice/issues/279
		// TRICE0() works in either case
		#ifdef __STRICT_ANSI__
		// if compiled with e.g. --std=c99
```
when running `renewIDs_in_examples_and_test_folder.sh`. This needs a special test first.

rokath, 2024-12-18:
I propose to close this issue for less issue noise. Normal usage is `trice(...` and all these cases are evaluated for matching value counts. If a user writes `TRICE0(...` or `TRICE_3(...` incorrectly, at least the compiler will complain. Also this limitation is documented now.

rokath, 2025-01-19:
Will be opened on demand again.

closed 2025-01-19 by rokath