rokath/trice #468: Help understanding triceRingBufferDepthMax diagnostic

KammutierSpule, 2024-05-27:
The following logs shows a Trice0Message and then various Trice( messages based on diagnostic.
TriceWordCounter and TriceOverflowCounter are my implementations.
All variables (except TriceOverflowCounter) are zeroed on each diagnostic report.

```
: file.c    40       25_696 Trice0Message
:     TriceDiagnostic.c   123       25_796 TriceWordCounter = 17, TriceOverflowCounter = 0
:     TriceDiagnostic.c   127       25_796 triceSingleDepthMax =  32 of 120, SingleTricesRingCountMax = 4, triceRingBufferDepthMax = 124 of  600
:     TriceDiagnostic.c   123       35_797 TriceWordCounter = 15, TriceOverflowCounter = 0
:     TriceDiagnostic.c   127       35_797 triceSingleDepthMax =  32 of 120, SingleTricesRingCountMax = 4, triceRingBufferDepthMax = 216 of  600

```
Would be possible to clarify the triceRingBufferDepthMax ? 
I can't understand how these values match with the other values (eg: TriceWordCounter and triceSingleDepthMax vs SingleTricesRingCountMax)
I can't make any calculation that will match.

TriceWordCounter is:
```
#define TRICE_DIAGNOSTICS_SINGLE_BUFFER do { \
    unsigned wordCount = TriceBufferWritePosition - triceSingleBufferStartWritePosition; \
    TriceWordCounter += wordCount; \
    TriceSingleMaxWordCount = (wordCount < TriceSingleMaxWordCount) ? TriceSingleMaxWordCount : wordCount; \
    } while(0);
```

KammutierSpule, 2024-05-27:
hold on... closing for better investigation.. :)

rokath, 2024-06-01:
Please check *triceRingBuffer.c* function `TriceEnoughSpace` in experimental branch now. It contains additional comments and a bug fix now. Could still not test it.

rokath, 2024-07-16:
The changes are in main branch now. There is now a switch TRICE_PROTECT, defaulting to 1 and it protects against buffer overflows. Successfully tested in `./test/dblB_protect_de_tcobs_ua` and `./test/ringB_protect_de_tcobs_ua`. Please re-open this issue if there are remaining open questions.

closed 2024-07-16 by rokath