rokath/trice #479: Unable to see logs if time stamp function are not implemented / are dummy.

mmurty, 2024-07-04:
I am using an STM32G070 device and I am trying to setup trice to log data.
I am using RTT for data transfer using a JLINK. I have done the required config in the config file.
I just wanted to know if we do not implement functions TriceStamp16() & TriceStamp32() will the PC agent display logs?
Presently I am getting errors and I am unable to see the logs (on a Windows machine).

I am running it using the command line and I see that it establishes connection with the JLINK correctly but it shows cycle count incorrect errors. Following is the command that I am using:

trice l -p TCP4 -args localhost:19021 -pf none -ts ""

rokath, 2024-07-04:
You could `#define TriceStamp16() 0` and `#define TriceStamp32() 0` to make the compiler happy. Alternatively set all `#define ENABLE_T[r|R]ice[8|16|32|64]fn_n  0` in *triceConfig.h*. In the target code only use `trice( "...\n");` and avoid `Trice( "...\n");`  and `TRice( "...\n");`. For RTT without framing the `-d16` switch is recommended just in case you accidentally write `Trice( "Hi\n");` If you still face problems please provide your *triceConfig.h* file together with the used command line.

> I just wanted to know if we do not implement functions TriceStamp16() & TriceStamp32() will the PC agent display logs?

The Trice tool does not need target stamps. The `-ts` switch is only relevant for the output formatting.

mmurty, 2024-07-11:
[trice_rtt_test.zip](https://github.com/user-attachments/files/16179092/trice_rtt_test.zip)

I have attached my project for your reference. You can check the triceConfig.h file.

I am getting this in the terminal:

[image]

I am using the latest trice version V0.63.0.
Following is the command I am using: `trice l -p TCP4 -args localhost:19021 -pf none -d16`

I apologize for late reply.

rokath, 2024-07-11:
If I understand it right, you expect every 1000 ms a log. This is not the case always. The time interval can be 2 or 3 seconds too. For example between "Hello World 25" and "Hello World 26" are 2 seconds and between "Hello World 16" and "Hello World 18" is only 1 second. This seems to be a timing issue in your setup. Please tell more about your hardware. Just for investigation I recommend temporary to use target timestamps. You can use the `-s` and `-v` switches also to see more information.

mmurty, 2024-07-12:
I understand that the host time stamps can be different based on when the RTT buffer is getting flushed. I am just not sure why the `CYCLE: XXX not equal expected value YYY - adjusting` errors are popping up? Does it indicate that I am missing data? If you observe the numbers in the `Hello World` message it seems that there are some misses.
In the past I have used target timestamps over UART on cortex M4 & M7 devices and everything went fine. This is first time I am trying it over RTT and unfortunately I do not have the ARM cortex cycle counter on this device to generate timestamps. I do not want to overload the CPU by generating periodic interrupts and hence I am not generating time stamps using a timer. 

My Hardware: 
MCU = STM32G070CBT6 ARM Cortex M0+, operates at 64MHz.
Debug Interface = SWD.
Transfer Lib = Segger RTT.
Debug Probe= Jlink Edu
Gdb Server= Started using Ozone debug session.
OS= Windows 11
Shell = Windows CMD
Target(MCU) Code Trice Version: V0.63.0
HOST(PC) Code Trice Version: V0.62.0 - Using pre-built binaries from the release (This is old version I will update and check it once).

rokath, 2024-07-12:
Yes, there seem to be data losses - therefore the `-s` could help. 1000ms is a really long time for a PC - so no doubt that there is something wrong. You could `#define TriceStamp32() (uwTick)` in *triceConfig.h* and use `TRice( "hello world %d\n, i" );` instead.

What is also possible: `#define TriceStamp32() (0x32323232)` in *triceConfig.h*. Then send a log with `-s`. 

I am not 100% sure about your TIM6 period. Is is guarantied to be 1ms? You could toggle a LED every 1000 times period to check that.

I never tried Ozone. May be the problem is there?

You could try to use the command line `trice log -p jlink -args="-Device STM32G070CB -if SWD -Speed 4000 -RTTChannel 0" -v`. The `-v` only for the setup checks. 

The STM Eclipse IDE is not my favorite one. You could consider to use vsCode with the Cortex-Debug extension just to get started

mmurty, 2024-07-12:
Using this command to test:
`trice log -p jlink -args="-Device STM32G070CB -if SWD -Speed 4000 -RTTChannel 0" -pf none`
and it works. Thanks for the suggestion. It seems when I am running the GDB server using any other app, some logs may be going missing and that's creating some issues.
This is the current output:
[image]

Do you want me to send any other logs just to make the tool better?

rokath, 2024-07-12:
If you would figure out, how the Ozone solution is to use in detail, that others could use it, that would be fine. Otherwise please simply close this issue.

mmurty, 2024-07-13:
Ozone has a terminal window in it where it shows RTT output. It must be causing some problems.

closed 2024-07-13 by mmurty