rokath/trice #494: Add support for %lu and other % formats

KammutierSpule, 2024-09-05:
It looks trice tool does not print correctly %lu

I believe this won't be the same on all platforms but should be investigated.

Or could you support %lu same as %u at least? so it still prints ok?

rokath, 2024-09-05:
All current possibilities are pointed out in [TriceVsPrintfSimilaritiesAndDifferences.md](https://github.com/rokath/trice/blob/master/docs/TriceVsPrintfSimilaritiesAndDifferences.md). In fact, the format strings are processed in Go and this language does not use `%lu` for example. It does even not support `%u`. To make `%u` possible, I exchange `%u` like occurences with `%d` counterparts and perform appropriate type casts. To get all the `%[0-9]l...` stuff working as well, the exchange technique is extendable. You could help by providing a reference list. Maybe it is easy: Just exchange `%[0-9]l...` with `%[0-9]...` without the need of type casts.

KammutierSpule, 2024-09-05:
to clarify, the hard to change on application side, is because some compilers by default (eg: ESP32 platform) they fail to compile when the % does not match the type of variable.

rokath, 2024-09-05:
Could you please provide an example showing that in detail? Afaik, the compilers do not see the format strings at all, because these are not compiled into the target image. All trice64 values are casted to `uint64_t` for example.

KammutierSpule, 2024-09-05:
> to clarify, the hard to change on application side, is because some compilers by default (eg: ESP32 platform) they fail to compile when the % does not match the type of variable.

Correction and Update: for Trice use, there is no issue, you may want to close this issue now if you wish.

```
TRice( iD(4530), "a_retry_timeout_ms:%u\n", a_retry_timeout_ms); // Works
printf("a_retry_timeout_ms:%u\n", a_retry_timeout_ms); // Build error
```

`
error: format '%u' expects argument of type 'unsigned int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  422 |     printf("a_retry_timeout_ms:%u\n", a_retry_timeout_ms);
      |                                ~^     ~~~~~~~~~~~~~~~~~~
      |                                 |     |
      |                                 |     uint32_t {aka long unsigned int}
      |                                 unsigned int
      |                                %lu
`

I was just mirroring the compiler checks for "printf" function types and using the same on trice.
but I don't need to.

rokath, 2024-09-05:
If it make sense to extend the documentation here in some way, please reopen this issue.

closed 2024-09-05 by rokath