rokath/trice #497: Proposal: Endianness on computer side

Sazerac4, 2024-09-09:
I saw that the embedded code can potentially change the endianess if needed before send data.

```c
	//! TRICE_HTOTS reorders short values from host order into trice transfer order.
	#define TRICE_HTOTS(x) Reverse16(x) // __bswap_16((x))) // TRICE_SWAPINT16(x)

	//! TRICE_HTOTL reorders long values from host order x into trice transfer order.
	#define TRICE_HTOTL(x) Reverse32(x) // __bswap_32((x))) // TRICE_SWAPINT32(x)

	//! TRICE_TTOHS reorders short values from trice transfer order into host order.
	#define TRICE_TTOHS(x) Reverse16(x) // __bswap_16((x))) // TRICE_SWAPINT16(x)
```

To allow to log as quickly as possible in most case,  can we not save on the target of endian management.
Isn't it up to the computer to perform the conversion operations ? (with `trice` app ) I guess it is needed to find a way to tell the utility what the endianness of the target is.

Maybe I'm missing context. When would endianness management on the target be useful ?

rokath, 2024-09-09:
There is a CLI switch:
```b
-triceEndianness string
    	Target endianness trice data stream. Option: "bigEndian". (default "littleEndian")
```
The user can set it to `bigEndian` for big endian MCUs to avoid conversion on the target side. The target macros are available if a use case needs network order or in a target network different endianness exists.

rokath, 2024-09-09:
You propose to drop the conversion macros? They are empty in most cases but can we guarantee, that nobody ever needs them?

To interpret the trice data stream only the first 2 bytes are to read as uint16_t in the known endianness. The 2 most significant bits code the following data. If they are both 0, a user protocol is possible. See [binary encoding](https://github.com/rokath/trice/blob/master/docs/TriceUserGuide.md#16-binary-encoding).

rokath, 2024-09-09:
> For networking, the trice output should not be an opaque sequence of bytes already encoded ?
Maybe I'm missing something.

For networking there should be a transport layer covering the Trice data. These maybe then should be in network order. In any case a conversion could happen deferred as well. 

I am not sure about the SEGGER_RTT when target endianness does not match host endianness.

Sazerac4, 2024-09-10:
> You propose to drop the conversion macros? They are empty in most cases but can we guarantee, that nobody ever needs them?

Yes. I'm curious if this can be guaranteed.

> For networking there should be a transport layer covering the Trice data. These maybe then should be in network order. In any case a conversion could happen deferred as well.

The transport layer is implemented by the user ? Or trice will do it ?

> I am not sure about the SEGGER_RTT when target endianness does not match host endianness.

I would like to test the RTT and see what happens.

rokath, 2024-09-12:
Maybe the `TRICE_HTOTS(x)`  like macros go as empty hooks into [triceDefaultConfig.h](https://github.com/rokath/trice/blob/master/src/triceDefaultConfig.h)?

rokath, 2024-09-17:
In the `refactor` branch now the endianness is handled completely. There have been some bugs in the newer code. There is a plan to refactor the code more (and also to optimize some parts) to finally have the option `TRICE_TRANSFER_ORDER_IS_NOT_MCU_ENDIAN == 1` available but without disturbing code reads too much. There is now also a test, which checks the `TRICE_TRANSFER_ORDER_IS_NOT_MCU_ENDIAN == 1` code parts. The advantage with `./test/be_dblB_de_tcobs_ua/` is, that it tests the Trice CLI switch `-triceEndianness bigEndian` automatically. If it works out, that we do not need the `TRICE_TRANSFER_ORDER_IS_NOT_MCU_ENDIAN == 1` code at all on the target side, we will find a way to transfer it to the C test code completely.

Sazerac4, 2024-09-22:
Hello,
I was not able to test the RTT, I cannot find my Jlink...
I don't have much time this September. I take advantage of it to read the source code and understand the output format.
I read the source code and I have some questions :

- In the differences of the macros trice8, trice16, trice32 and trice64. Their activation depends on the maximum size of the arguments that we want: `TRICE_DEFAULT_PARAMETER_BIT_WIDTH`. But when writing to the buffer, data must always be format to fit a 32 bit size?

- What is the purprose to always put data in a 32bits format ? (target or memeory alignement ? )

- I understand the syntax of macros and the feature attached to them but not the full uppercase one :
```c
#define trice8(tid, fmt, ...)   // Log (no timestamp)
#define Trice8(tid, fmt, ...)  // Log (16 bits timestamp)
#define TRice8(tid, fmt, ...)  // Log (32 bits timestamp)
//
#define TRICE8(tid, fmt, ...)  // What is the purpose of this kind of macros ?
```
- When i read this macros (`trice8.h:93`). I think `tid` is unused. it is right guess ?

```c
//! TRICE8_0 writes trice data as fast as possible in a buffer.
//! \param tid is a 16 bit Trice id in upper 2 bytes of a 32 bit value
#define TRICE8_0(tid, pFmt) \
	TRICE_ENTER tid;        \
	TRICE_CNTC(0);          \
	TRICE_LEAVE
```

I think I'm a bit off topic here. Maybe an issue on the "refactor" branch would be better.

rokath, 2024-09-22:
> In the differences of the macros trice8, trice16, trice32 and trice64. Their activation depends on the maximum size of the arguments that we want: TRICE_DEFAULT_PARAMETER_BIT_WIDTH. But when writing to the buffer, data must always be format to fit a 32 bit size?

The macros trice8, trice16, trice32 and trice64 are always usable and the number 8, 16, 32, 64 specifies the parameter width. They are partially disabled, when the value `TRICE_SINGLE_MAX_SIZE` is defined to be smaller than 104. That's mainly to get compiler errors rather than runtime errors. The value `TRICE_DEFAULT_PARAMETER_BIT_WIDTH` is the parameter bit with for the macro `trice` (without number). It can make sense to set this value to 16 on smaller machines. 

> What is the purprose to always put data in a 32bits format ? (target or memeory alignement ? )

Mainly speed. If you do all byte, wise things slow down. Of course alignment is important when using 32-bit instructions. The Trice macro base size is 4 bytes. If, for example a single 8-bit value is parameter, its includes 4 padding 0-bytes. These are not transferred - they only occupy RAM. A 64-bit alignment would be overkill, so 32-bit seems to be a good choice, as many MCUs nowadays are 32-bit ones.

> I understand the syntax of macros and the feature attached to them but not the full uppercase one

"the full uppercase one" is the real Trice macro only using inline code. Because the main design aim was speed, this was the original design. Then I realized, that several hundred of `TRICE` macros increase the needed code amount too much and that is better to have just a function call instead of having inline macros. If speed matters use `TRICE(id(0)`, `TRICE(Id(0)`, `TRICE(ID(0)` else use `trice(iD(0)`, `Trice(iD(0)`, `TRice(iD(0)` or mix usage as you like. They internally use `TRICE` but each is only a function call then.

The `TRICE` macro definition uses the C preprocess to concatenate the code behind the sub-macros 
`TRICE_ENTER`
 tid=`id(n)`, `Id(n)`, `ID(n)` 
`TRICE_CNTC(0)`
`TRICE_LEAVE`

> When i read this macros (trice8.h:93). I think tid is unused. it is right guess ?

_tid_  in this context is executable code `id`, `Id` or `ID` defined in trice.h:

```C
//! ID writes 14-bit id with 11 as 2 most significant bits, followed by a 32-bit stamp.
//! 11iiiiiiI TT | TT (NC) | ...
//! C000 = 1100 0000 0000 0000
#define ID(n)                                 \
	do {                                      \
		uint32_t ts = TriceStamp32;           \
		TRICE_PUT16_1616((0xC000 | (n)), ts); \
	} while (0)

//! Id writes 14-bit id with 10 as 2 most significant bits two times, followed by a 16-bit stamp.
//! 10iiiiiiI 10iiiiiiI | TT (NC) | ...
//! 8000 = 1000 0000 0000 0000
#define Id(n)                                        \
	do {                                             \
		uint16_t ts = TriceStamp16;                  \
		TRICE_PUT((0x80008000 | ((n) << 16) | (n))); \
		TRICE_PUT16(ts);                             \
	} while (0)

//! id writes 14-bit id with 01 as 2 most significant bits, followed by no stamp.
//! 01iiiiiiI (NC) | ...
//! 4000 = 0100 0000 0000 0000
#define id(n) TRICE_PUT16(0x4000 | (n));

//! iD is just a code parsing helper.
#define iD(n) (n)
```

rokath, 2024-09-23:
> I think I'm a bit off topic here. Maybe an issue on the "refactor" branch would be better.

What about tranferring this part to a discussion?

rokath, 2024-09-28:
The endianness branch now is in a good shape from my point of view and all current tests are okay. The byte swapping code is now more concentrated. If you do not have any critical remarks on it, I would merge it to the master branch within the next days.
Unfortunately there are still 2 locations for `TRICE_REVERSE`. Right now I do not know how to get around it easily.

rokath, 2024-09-30:
All byte order dependent code is now at one point only in the endianness branch and `TRICE_REVERSE` is not needed anymore:

```C
#if TRICE_TRANSFER_ORDER_IS_BIG_ENDIAN == TRICE_MCU_IS_BIG_ENDIAN

// TRICE_REVERSE == 0 uses no byte swapping inside the Trice macros resulting in less code and faster execution.
#include "triceMcuOrder.h"
#include "trice8McuOrder.h"
#include "trice16McuOrder.h"
#include "trice32McuOrder.h"
#include "trice64McuOrder.h"

#else // #if TRICE_TRANSFER_ORDER_IS_BIG_ENDIAN == TRICE_MCU_IS_BIG_ENDIAN

// TRICE_REVERSE == 1 causes byte swapping inside the Trice macros resulting in more code and slower execution.
#include "triceMcuReverse.h"
#include "trice8McuReverse.h"
#include "trice16McuReverse.h"
#include "trice32McuReverse.h"
#include "trice64McuReverse.h"

#endif // #else // #if TRICE_TRANSFER_ORDER_IS_BIG_ENDIAN == TRICE_MCU_IS_BIG_ENDIAN
```

rokath, 2024-10-01:
Merged into master

rokath, 2024-10-03:
The branch is now merged into master. 

>Perhaps we can add after the "buildin" instructions when the compiler used is detected.
For GCC or clang, we can use: __builtin_bswap64 __builtin_bswap32 __builtin_bswap16

The `buildin_`... are of course a better approach.

rokath, 2024-10-08:
After https://github.com/rokath/trice/pull/505: Should we close this issue now?

Sazerac4, 2024-10-09:
Ok for me

closed 2024-10-09 by rokath