rokath/trice #509: Fail to compile with TRICE_OFF=1

KammutierSpule, 2024-11-02:
When I build with TRICE_OFF=1 set globally to the project (compiler) I got these errors:

```
error: 'TRice0' was not declared in this scope;
...
error: 'TRice32' was not declared in this scope
```

KammutierSpule, 2024-11-02:
on triceAuxiliary.c
does #if TRICE_DEFERRED_AUXILIARY8 == 1
and others,
missing && TRICE_OFF == 0 ?

KammutierSpule, 2024-11-02:
> ```
> error: 'TRice0' was not declared in this scope;
> error: 'TRice32' was not declared in this scope
> ```

I believe this is due the 
```
#if ((defined(TRICE_OFF) && TRICE_OFF == 1)) || ((defined(TRICE_CLEAN) && TRICE_CLEAN == 1))
#include "triceOff.h" <-- missing here TRice0 here
#else // #if ((defined(TRICE_OFF) && TRICE_OFF == 1)) || ((defined(TRICE_CLEAN) && TRICE_CLEAN == 1))
..
```
They are added here, but at this point, the code here is not reached because of the previous defines.
```
#if TRICE_OFF == 1 || TRICE_CLEAN == 1
// clang-format off
	TRICE_INLINE void trice0( const char * pFmt ){TRICE_UNUSED(pFmt)}
	TRICE_INLINE void Trice0( const char * pFmt ){TRICE_UNUSED(pFmt)}
	TRICE_INLINE void TRice0( const char * pFmt ){TRICE_UNUSED(pFmt)}
```
so should this move to triceOff?

rokath, 2024-11-03:
Thanks for this information. Would you please provide your `triceConfig.h` and, if relevant, your Makefile.

KammutierSpule, 2024-11-03:
triceConfig.h
I think the most relevant part,

```
#define TRICE_DEFERRED_AUXILIARY8 1
#define TRICE_BUFFER TRICE_RING_BUFFER
#define TRICE_DEFERRED_TRANSFER_MODE TRICE_SINGLE_PACK_MODE
#define TRICE_DIRECT_OUTPUT 0
#define TRICE_DEFERRED_OUTPUT 1
#define TRICE_DATA_OFFSET 64
#define TRICE_SINGLE_MAX_SIZE 104  // must be a multiple of 4
#define TRICE_DEFERRED_BUFFER_SIZE 1024  // must be a multiple of 4
#define TRICE_MCU_IS_BIG_ENDIAN 0
#define TRICE_TRANSFER_ORDER_IS_NOT_MCU_ENDIAN 0
#define TRICE_DEFERRED_OUT_FRAMING TRICE_FRAMING_TCOBS
#define TRICE_DEFERRED_XTEA_ENCRYPT 0
#define TRICE_DIAGNOSTICS 1
```

Can't share something useful regarding the Makefile (actually, cmakelist.txt), but it just sets TRICE_OFF=1 globally.

rokath, 2024-11-04:
Hopefully fixed in `devel` branch. I could not reproduce the `error: 'TRice32' was not declared in this scope` message. Please check if it is ok now and close this issue then. Otherwise please give more details. BTW: You do not need to define default values in `triceConfig.h`. But to make sure, things are as expected this is ok.

KammutierSpule, 2024-11-04:
I got these new errors now, is this something?
```
error: macro "TRice32" requires 3 arguments, but only 1 given
error: 'TRice32' was not declared in this scope
```

rokath, 2024-11-04:
All tests run successfully and also the examples are translatable with `-DTRICE_OFF=1`. The examples are including `triceCheck.c`, which also has `TRice32(...` usages. I have right now no idea how to proceed other, than asking you to provide me some more information so that I can reproduce the problem or that you try to find out the root cause yourself.

rokath, 2024-11-04:
One option could me to define the following block as empty macros, when `TRICE_OFF == 1`. But I first would like to understand why I am not able to produce the error.

KammutierSpule, 2024-11-04:
Another new error I'm getting, on a file with:

```
#define TRICE_OFF 1
#include <trice.h>

error: 'iD' was not declared in this scope
```

again, looks related to
```
#if ((defined(TRICE_OFF) && TRICE_OFF == 1)) || ((defined(TRICE_CLEAN) && TRICE_CLEAN == 1))
#include "triceOff.h"

#else // #if ((defined(TRICE_OFF) && TRICE_OFF == 1)) || ((defined(TRICE_CLEAN) && TRICE_CLEAN == 1))
<<<-- ids are declared only here
```

rokath, 2024-11-04:
If you pull the branch devel, can you run examples/G01B_inst/build.sh TRICE_OFF=1 successful? What differs to your Project?

rokath, 2024-11-04:
OK, with this I could reproduce the problem:

```C
  TRice("w: Hello! 👋🙂 \a\n" ); // with sound!
  TRice0("w: Hello! 👋🙂 \a\n" ); // with sound!
  TRice8("w: Hello! 👋🙂 \a\n" ); // with sound!
  TRice16("w: Hello! 👋🙂 \a\n" ); // with sound!
  TRice32("w: Hello! 👋🙂 \a\n" ); // with sound!
  TRice64("w: Hello! 👋🙂 \a\n" ); // with sound!
  Trice("w: Hello! 👋🙂 \a\n" ); // with sound!
  Trice0("w: Hello! 👋🙂 \a\n" ); // with sound!
  Trice8("w: Hello! 👋🙂 \a\n" ); // with sound!
  Trice16("w: Hello! 👋🙂 \a\n" ); // with sound!
  Trice32("w: Hello! 👋🙂 \a\n" ); // with sound!
  Trice64("w: Hello! 👋🙂 \a\n" ); // with sound!
  trice("w: Hello! 👋🙂 \a\n" ); // with sound!
  trice0("w: Hello! 👋🙂 \a\n" ); // with sound!
  trice8("w: Hello! 👋🙂 \a\n" ); // with sound!
  trice16("w: Hello! 👋🙂 \a\n" ); // with sound!
  trice32("w: Hello! 👋🙂 \a\n" ); // with sound!
  trice64("w: Hello! 👋🙂 \a\n" ); // with sound!
```
I will check tomorrow...

KammutierSpule, 2024-11-05:
Looks good now, thanks!

closed 2024-11-05 by KammutierSpule