rokath/trice #510: trice clean inconsistency

KammutierSpule, 2024-11-23:
I'm experience running trice tool on different containers, somehow, they "clean differently"
In one system it cleans to `Trice(<space>"....`
on other system I got the <space> removed `Trice("....`

I'm not sure what are the differences on my system that may be causing this.

rokath, 2024-11-23:
Are all used Trice tool versions are identical? There is a switch `-space` for `trice insert` but not for `trice clean` what may be an error.

rokath, 2024-11-24:
The switch `-spacing` is now renamed into `-spaceAfterOpeningBrace`, or short `-w`, and exists now for `trice i` and `trice c` in branch devel.

KammutierSpule, 2024-11-25:
thanks! Can you clarify what was the previous default behavior of insert and clean, and now what is the default behavior?

KammutierSpule, 2024-11-25:
Thanks I will evaluate later.
My main concern is more with version control system, as if on different systems it has different behavior it will cause a big diff on files.

rokath, 2024-11-25:
If the Trice version is the same on different systems there should be no
differences even if compiled with different Go versions. Otherwise this
would be very interesting.

rokath, 2024-12-01:
I wrote

> I am not sure, if the iD(n), should be optional also be iD(<space>n<space>),. We should keep it simple.

After thinking and reviewing the code I decided, that iD(n) should be influenced be the `-w` CLI switch as well. It was a very small change and is in my opinion now more consistent. The parser tolerates white spaces anyway. The long name of `-w` is now `-spaceInsideParenthesis`, no better idea.

rokath, 2024-12-12:
Can we close this or do you need more time?

KammutierSpule, 2024-12-12:
You can close, I didn't get a chance to test, but if I experience it again I will open a net issue with more detailed tests and information