rokath/trice #515: #define TRICE_CLEAN 1 results in compiler error triceStaticBuffer.c:9:28: error: 'TRICE_BUFFER_SIZE' undeclared here

initdb, 2025-01-10:
**Describe the bug**
I updated from v0.62.1 to v72.5. I build trice as a static lib an link it into my project. I wanted to enable TRICE_CLEAN,
but this results in a compiler error:
```
/tmp/cs-trice/trice/src/triceStaticBuffer.c:9:28: error: 'TRICE_BUFFER_SIZE' undeclared here (not in a function); did you mean 'TRICE_RING_BUFFER_SIZE'?
    9 | uint32_t triceSingleBuffer[TRICE_BUFFER_SIZE >> 2];
      |                            ^~~~~~~~~~~~~~~~~
      |                            TRICE_RING_BUFFER_SIZE
/tmp/cs-trice/trice/src/triceStaticBuffer.c:14:24: error: macro "TriceTransfer" passed 1 arguments, but takes just 0
   14 | void TriceTransfer(void) {}
      |                        ^
In file included from /tmp/cs-trice/trice/src/trice.h:151,
                 from /tmp/cs-trice/trice/src/triceStaticBuffer.c:4:
/tmp/cs-trice/trice/src/triceOff.h:21: note: macro "TriceTransfer" defined here
   21 | #define TriceTransfer()
      | 
/tmp/cs-trice/trice/src/triceStaticBuffer.c:14:26: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
   14 | void TriceTransfer(void) {}
      |                          ^
```

it seems like the block which is defining TRICE_BUFFER_SIZE or my triceConfig.h is not compatible since the new examples are way shorter and offer less explanation.

**To Reproduce**
Steps to reproduce the behavior:

1. see my triceConfig.h
```c
/*! \file triceConfig.h
\author Thomas.Hoehenleitner [at] seerose.net
*******************************************************************************/

#ifndef TRICE_CONFIG_H_
#define TRICE_CONFIG_H_

#ifdef __cplusplus
extern "C" {
#endif

#include <stdint.h>
#include "timedate.h"

//! TRICE_CLEAN, if found inside triceConfig.h, is modified by the Trice tool to silent editor warnings in the cleaned state.
#define TRICE_CLEAN 1 // Do not define this at an other place! But you can delete this here.

//! TriceStamp16 returns a 16-bit value to stamp `Id` TRICE macros. Usually it is a timestamp, but could also be a destination address or a counter for example.
//! The user has to provide this function. Defining a macro here, instead if providing `int16_t TriceStamp16( void );` has significant speed impact.
#define TriceStamp16 timedate_get16bittime()

//! TriceStamp32 returns a 32-bit value to stamp `ID` TRICE macros. Usually it is a timestamp, but could also be a destination address or a counter for example.
//! The user has to provide this function. Defining a macro here, instead if providing `int32_t TriceStamp32( void );` has significant speed impact.
#define TriceStamp32 timedate_get32bittime()

//!  TRICE_BUFFER selects, where the TRICE macros accumulate the trice data during a single TRICE execution. Selectable options:
//! - TRICE_STACK_BUFFER: No additional buffer is needed, what makes sense for single task systems with direct output only.
//! - TRICE_STATIC_BUFFER: A single trice is stored in a separate static buffer, what makes sense in single- and multi-tasking systems with direct output only.
//! - TRICE_DOUBLE_BUFFER: TRICE macros write direct into a double buffer without any additional management action.
//!   This is the fastest execution option for TRICE macros but needs more RAM. Used for deferred output and optional additional direct output.
//! - TRICE_RING_BUFFER: TRICE macros write direct into a ring buffer without any additional management action.
//!   This is not the fastest execution option for TRICE macros but needs less RAM. Used for deferred output and optional additional direct output.
//! If unsure select TRICE_DOUBLE_BUFFER. The TRICE_RING_BUFFER option works, but is experimental.
#define TRICE_BUFFER TRICE_STATIC_BUFFER

//! TRICE_DIRECT_OUTPUT == 0: only deferred output, usually UART output only
//! TRICE_DIRECT_OUTPUT == 1: with direct output, usually RTT
//! Setting TRICE_BUFFER to TRICE_STACK_BUFFER or TRICE_STATIC_BUFFER demands TRICE_DIRECT_OUTPUT == 1, no deferred output at all.
//! When TRICE_BUFFER == TRICE_RING_BUFFER or TRICE_BUFFER == TRICE_DOUBLE_BUFFER for deferred output, additional direct output can be switched on here.
//! For example it is possible to have direct 32-bit wise RTT TRICE_FRAMING_NONE output and deferred UART TRICE_FRAMING_COBS output.
//! TRICE_BUFFER == TRICE_STACK_BUFFER or TRICE_BUFFER == TRICE_STATIC_BUFFER needs TRICE_DIRECT_OUTPUT == 1.
#define TRICE_DIRECT_OUTPUT 1
[... 289 lines cut ...]
#define ENABLE_TRice64fn_12 1

#endif // #if TRICE_32_BIT_SUPPORT

//
///////////////////////////////////////////////////////////////////////////////

#ifdef __cplusplus
}
#endif

#endif /* TRICE_CONFIG_H_ */
```

2. compile
3. get errors as stated above

**Expected behavior**
i would expect trice to compile, is there a detailed triceConfig.h template available like i used? Or are the triceConfig.h now just simplified? 

**Desktop (please complete the following information):**

- OS: ubuntu
- compiler: arm-none-eabi-gcc
- Version 12

rokath, 2025-01-10:
Please check out ./src/tricedefaultconfig.h. It contains all default definitions with documentation. You may need to rebuild your trice library.

initdb, 2025-01-16:
I will remove all defaults from the triceConfig.h.

One question: does the trice tool need to be run first before building trice as a lib if TRICE_CLEAN is enabled?

rokath, 2025-01-16:
What Trice does, is insert and clean IDs in source code. The Trice code itself does not have any IDs. So when building your Trice library only with Trice code from the ./src folder a Trice run is not needed. But if you include any other files containing Trice statements, a Trice run is mandatory, when you want to edit your code without inserted IDs. Please do not deal with TRICE_CLEAN. Only insert `#define TRICE_CLEAN 1` into your project specific triceConfig.h and only there. The Trice tool will automatically switch the value between 0 and 1.

initdb, 2025-01-16:
thanks for the quick response!

so the triceCongig.h is not needed for compiling trice as a lib? in this case i would just add the triceConfig.h to my project or is the triceConfig.h needed to compile the trice lib but the `#define TRICE_CLEAN 1`  has to be added to triceConfig.h and is just used by the trice tool. 

my main intention to activating `#define TRICE_CLEAN 1` are all the compiler warnings in my project for every source file which includes trice.h e.g.

`
/project/external/include/trice/trice.h: In function 'Trice0':
/project/external/include/trice/trice.h:875:54: warning: unused parameter 'pFmt' [-Wunused-parameter]
  875 | TRICE_INLINE void Trice0( uint16_t tid, char const * pFmt ){ Trice32m_0( tid ); }
      |                                         ~~~~~~~~~~~~~^~~~
`

rokath, 2025-01-17:
The triceConfig.h is mandatory for the trice code. It controls which parts of the trice code are included. I see no big advantage having a trice library, because it would work only with unchanged settings in the project specific triceConfig.h. Once the trice source files are translated, their objects are rebuilt automatically and only when the thriceConfig.h is changed. So only the linker has a bit less to do when it finds a trice library compared to a bunch of trice objects. But does that influence the build time heavily?

The triceConfig.h is the only part of the trice sources which should be modified by the users. It is ment to be a individual part of the user projects. The examples folder shows the usage.

TRICE_CLEAN==1 changes all Trice macros into empty ones. It is used only to silence sophisticated editors. In the cleaned state, when the IDs are removed from the files, the editor could underline the Trice macros indicating a false positive.

If you want to disable Trice macros use TRICE_OFF.

 Look in the TriceUserManual.md too.

rokath, 2025-01-17:
Concerning the usage of Trice0 you need to check the compiler version / mode. Please search for older issues. You may need to apply the `-c99` flag.

initdb, 2025-01-22:
> The triceConfig.h is mandatory for the trice code. It controls which parts of the trice code are included. I see no big advantage having a trice library, because it would work only with unchanged [...]

the mandatory triceConfig.h is clear. I'll update mine like suggested. 
I handle all dependencies by having them as an extra repo with a custom pipeline script which builds a package and pushes it to a configuration management repo. this way all dependencies are decupled and it is possible to have multiple variants of a dependency.

on my setup (windows + toolchain in docker container) having dependencies as static libs heavily improves the build time. since docker on windows uses the wsl2 filesystem which is increadably slow when handling many small files like object files.

PS: i promised a CMake file to build trice last year, its i think one refactoring away from beeing good. I will send it once I'm ready.

> The triceConfig.h is the only part of the trice sources which should be modified by the users. It is ment to be a individual part of the user projects. The examples folder shows the usage.

yes, but the template/ examples are now quite shorter than when i integrated trice.

> TRICE_CLEAN==1 changes all Trice macros into empty ones. It is used only to silence sophisticated editors. In the cleaned state, when the IDs are removed from the files, the editor could underline [...]

yeah CLion complains, but i have a pre build actions which runs trice before compilation, so i quess no need on my end.

> Concerning the usage of Trice0 you need to check the compiler version / mode. You may need NOT to apply the -c99 flag. See for > more information https://github.com/rokath/trice/issues/279 and https://github.com/rokath/trice/issues/432. [edited]

thanks, i will check if the warning labels go away with the thinned triceConfig.h. I'm using C11, but i will have a look.

initdb, 2025-01-22:
either the config change or the update removed all the Trice0 compiler warnings :) so defenetly a layer 8 problem.
i still have one with TRICE_S and TRICE_N, could this be because it has to use ID() instead of iD()?

```
external/include/trice/trice.h:707:15: warning: statement with no effect [-Wunused-value]
17:07:44     707 | #define iD(n) (n)
17:07:44         |               ^
17:07:44   external/include/trice/trice.h:616:29: note: in definition of macro 'TRICE_N'
17:07:44     616 |                 TRICE_ENTER tid;                                                                                                             \
17:07:44         |                             ^~~
17:07:44  source.c:678:5: note: in expansion of macro 'TRICE_S'
17:07:44     678 |     TRICE_S(iD(6500), "INFO:  connect to %s\n", buf);
17:07:44         |     ^~~~~~~
17:07:44   source.c:678:13: note: in expansion of macro 'iD'
17:07:44     678 |     TRICE_S(iD(6500), "INFO: connect to %s\n", buf);
```

rokath, 2025-01-23:
iD and ID are not exchangable.

I recommend to use triceN instead of TRICE_N for example. TRICE_N cannot deal with iD. It understands only id, Id and ID. The iD is reserved for lower case and camel case Trice macros. The TriceUserManua.md explains the details.

initdb, 2025-01-24:
thanks triceN, TriceN are newer additons? i think last time i added trice only TRICE_N was available?
now there a no more trice compiler warnings.

But one more question:
in the guide it says to run trice c after trice i. when i run trice c the ids get removed from the source files and then TRICE_CLEAN = 1 would come into play.
do i understand this correctly that if you do this without the cache the ids get changed completly, if i don't i get some sort of warning or what is this new print (I noticed with the most recent trice cli version.):

```
...
tag.idSpace= [1000 1001 1002 1003 1004 1005 1006 1007 1008 1009 1010 1011 1012 1013 7776 1015 1016 1017 1018 1019 1020 1021 1022 1023 7711 1025 1026 1027 1028 1029 1030 1031 1032 1033 1034 1035 7651 1037 1038 1039 1040 1041 1042 1043 1044 1045 1046 1047 1048 1049 1050 1051 1052 1053 1054 1055 1056 1057 1058 1059 1060 1061 1062 1063 1064 1065 1066 1067 7916 1069 1070 7627 1072 1073 1074 1075 1076 1077 1078 1079 1080 1081 1082 7913 1084 1085 1086 1087 1088 1089 1090 1091 1092 1093 1094 1095 1096 1097 1098 1099 1100 1101 1102 1103 1104 7619 1106 1107 1108 1109 1110 1111 1112 1113 1114 1115 1116 1117 1118 1119 1120 1121 1122 1123 1124 1125 1126 1127 1128 1129 1130 1131 1132 1133 1134 1135 7861 1137 1138 1139 1140 1141 1142 1143 1144 1145 1146 1147 1148 7799 1150 7710 1152 1153 1154 1155 1156 1157 1158 1159 1160 1161 1162 1163 7637 1165 1166 1167 1168 7747 1170 1171 1172 1173 1174 1175 1176 1177 1178 1179 1180 1181 1182 1183 1184 1185 1186 1187 1188 1189 1190 1191 1192 1193 7941 1195 1196 1197 1198 1199 1200 1201 1202 1203 1204 1205 1206 1207 1208 1209 1210 1211 1212 1213 1214 1215 1216 1217 1218 1219 1220 7552 1222 1223 1224 1225 1226 1227 1228 1229 1230 1231 1232 7971 7542 1235 1236 1237 1238 1239 1240 1241 1242 1243 1244 1245 1246 1247 1248 1249 1250 1251 1252 1253 1254 1255 1256 1257 1258 1259 1260 1261 1262 1263 1264 1265 1266 1267 1268 1269 1270 1271 1272 1273 1274 1275 1276 1277 1278 1279 1280 1281 1282 1283 1284 1285 1286 7613 1288 1289 1290 1291 1292 1293 1294 1295 1296 1297 1298 1299 1300 1301 7559 1303 1304 1305 1306 1307 1308 1309 1310 1311 1312 1313 1314 1315 1316 1317 1318 1319 1320 1321 1322 1323 1324 1325 1326 1327 7974 1329 1330 1331 7921 1333 1334 1335 1336 1337 1338 1339 1340 1341 1342 1343 1344 1345 1346 1347 1348 1349 1350 1351 1352 1353 1354 1355 7702 1357 1358 1359 1360 1361 1362 1363 1364 1365 1366 1367 1368 1369 1370 1371 1372 1373 1374 1375 1376 1377 1378 1379 1380 1381 1382 1383 1384 1385 1386 1387 1388 1389 1390 1391 1392 1393 1394 1395 7871 1397 1398 1399 1400 1401 1402 7668 1404 1405 1406 1407 1408 1409 1410 1411 1412 1413 1414 1415 1416 1417 1418 1419 1420 1421 1422 1423 1424 1425 1426 1427 1428 1429 1430 1431 1432 1433 1434 1435 1436 7653 1438 1439 1440 1441 1442 1443 1444 1445 1446 7614 1448 1449 1450 1451 1452 1453 1454 1455 1456 1457 1458 1459 1460 1461 1462 1463 1464 1465 1466 1467 1468 1469 1470 1471 1472 1473 1474 1475 1476 1477 1478 1479 1480 1481 1482 1483 1484 1485 7887 1487 1488 1489 1490 7719 1492 1493 1494 1495 1496 1497 1498 1499 1500 1501 1502 1503 1504 1505 1506 1507 1508 1509 1510 1511 1512 1513 1514 1515 1516 1517 1518 1519 1520 1521 1522 1523 1524 1525 1526 1527 1528 7769 1530 1531 1532 1533 1534 1535 1536 1537 1538 1539 1540 1541 1542 1543 7679 1545 1546 1547 1548 1549 1550 1551 1552 1553 7744 1555 1556 1557 1558 7573 7695 1561 1562 1563 1564 1565 1566 1567 7633 1569 1570 1571 1572 7777 1574 7772 1576 1577 1578 1579 1580 1581 1582 1583 1584 1585 1586 1587 1588 1589 1590 1591 1592 7922 1594 1595 1596 1597 1598 1599 1600 7954 1602 1603 1604 1605 1606 1607 1608 1609 1610 1611 1612 1613 1614 1615 1616 1617 1618 7853 1620 1621 1622 7659 1624 1625 1626 1627 1628 1629 1630 1631 1632 7746 1634 1635 1636 1637 7570 1639 1640 1641 1642 1643 1644 1645 1646 1647 1648 1649 1650 1651 1652 1653 1654 1655 1656 1657 1658 1659 1660 1661 1662 1663 1664 1665 1666 1667 1668 1669 1670 1671 1672 1673 1674 1675 1676 1677 1678 1679 1680 1681 1682 1683 1684 1685 1686 1687 7684 1689 1690 1691 1692 1693 1694 1695 7578 1697 1698 1699 1700 1701 1702 1703 1704 1705 1706 1707 1708 1709 1710 1711 1712 1713 1714 1715 1716 1717 1718 1719 1720 1721 1722 7588 1724 1725 1726 1727 1728 1729 1730 1731 7928 1733 1734 1735 1736 1737 1738 1739 1740 1741 1742 1743 1744 1745 1746 7751 7686 1749 1750 1751 1752 7523 1754 1755 1756 1757 1758 1759 1760 1761 1762 7875 1764 1765 1766 1767 1768 1769 1770 1771 1772 1773 1774 1775 1776 1777 1778 1779 1780 1781 1782 1783 1784 1785 1786 1787 1788 1789 1790 1791 1792 1793 1794 7582 1796 1797 1798 1799 7960 1801 1802 1803 1804 1805 1806 1807 1808 1809 1810 1811 1812 1813 1814 1815 1816 1817 1818 1819 1820 1821 1822 1823 1824 1825 1826 1827 1828 1829 1830 1831 1832 1833 1834 1835 1836 1837 1838 1839 1840 1841 1842 7842 1844 1845 1846 1847 1848 1849 1850 1851 1852 1853 1854 1855 1856 1857 1858 1859 1860 1861 7866 1863 1864 1865 1866 1867 1868 1869 1870 1871 1872 1873 1874 1875 1876 1877 1878 1879 1880 1881 1882 1883 1884 1885 1886 1887 1888 1889 1890 1891 1892 1893 1894 1895 1896 1897 1898 1899 1900 1901 1902 1903 1904 1905 1906 1907 1908 1909 1910 1911 1912 7676 1914 1915 1916 7877 1918 1919 1920 1921 1922 1923 1924 1925 1926 1927 1928 1929 1930 7730 1932 1933 1934 1935 1936 1937 1938 1939 1940 1941 1942 7664 1944 1945 1946 1947 1948 1949 1950 1951 1952 1953 1954 1955 1956 1957 1958 1959 7945 1961 1962 1963 1964 7715 1966 1967 1968 1969 1970 1971 1972 1973 1974 1975 1976 1977 1978 1979 1980 1981 1982 1983 1984 1985  [... cut ...] 23 7424 7942 7426 7427 7428 7429 7430 7431 7432 7433 7434 7435 7436 7437 7438 7754 7440 7441 7442 7443 7807 7445 7446 7447 7448 7449 7450 7451 7452 7453 7454 7455 7456 7457 7458 7459 7460 7461 7462 7463 7464 7465 7466 7467 7639 7469 7470 7471 7472 7473 7474 7475 7476 7477 7478 7616 7480 7481 7482 7483 7484 7485 7486 7487 7915 7996 7490 7491 7492 7493 7494 7495 7496 7497 7498 7499 7500 7501 7502 7503 7504 7505 7506 7507 7508 7509 7510 7511 7512 7513 7514 7515 7516 7517] idx= 3971 iD= 7838
...
```

rokath, 2025-01-24:
> thanks triceN, TriceN are newer additons? i think last time i added trice only TRICE_N was available?

Yes. The upper-case-only Trice macros  include Trice code instead of calling it. That has little speed improvement but needs more flash for the code especially when you have many Trice macros. 

> do i understand this correctly that if you do this without the cache the ids get changed completly, 

No. All IDs stay the same with or without the cache. The IDs are kept inside the til.json file. As long you do not delete it, your whole history is available. The sense of the Trice cache is, that `trice i` and `trice c` do not need to touch the source files when they are unchanged.  The files get simply exchanged with their counter parts in the Trice cache. That is faster than processing but the main advantage is, that the compiler do not need to re-translate. Just try it out by using the `-cache` switch or not.

initdb, 2025-01-24:
tanks for the quick response. Is there a disadvantage to not run trice c after trice i? e.g. the printing of the tag.idSpace= logs?

rokath, 2025-01-24:
> Is there a disadvantage to not run trice c after trice i? 

No need to run `trice c` at all. Then you need also no cache. The whole effort is only to make the source code better readable. With `trice c` in your build script you do not see the IDs in your source code anymore.

That's you see when using `trice c`:

```C
trice( "Hi!\n" );
```

If you prefer to see this:

```C
trice( iD(12345), "Hi!\n" );
```

Then do not use `trice c `.

initdb, 2025-01-24:
ok understood, im totally fine with the ids in the code. so no trice c and no cache. And the last part of trice i printing a lot of lines of tag.idSpace= ? this is also new. Excuse the many questions i just want to understand the tool a little better.

rokath, 2025-01-24:
>  im totally fine with the ids in the code. so no trice c and no cache

Then you can remove the `#define TRICE_CLEAN 0` from your triceConfig.h too. 

>  And the last part of trice i printing a lot of lines of tag.idSpace= ? 

Are you using the verbose flag `-v`? The Trice tool should work silently.

initdb, 2025-01-30:
thank you! the extra debug log is gone with 0.73.1.
Sadly i have a new issue.
now trice locks or gets stuck on the first trice message of the application.
I will create a new issue for this.

this issue can be closed from my end.

closed 2025-01-31 by rokath