rokath/trice #518: Trice locks up or gets stuck on first message of the application

initdb, 2025-01-30:
**Describe the bug**
I upgraded from 0.72.5 to 0.73.1 and now my application gets stuck on the first Trice message after TriceInit().

I marked the debugger step where it gets stuck with: 	=====>
I also will add the sybols tomorrow.

```c
//main:
TriceInit();

	=====>TRice(iD(7489), "INFO: APP_THREADX - init...!\n");

// trice.h
TRICE_INLINE void TRice0(uint16_t tid, const char* pFmt) {
	=====>TRice32m_0(tid);
	TRICE_UNUSED(pFmt)
}

// libtrice.a only dissassembly i will add the simbols tomorrow
TriceNonBlockingDirectWrite:
	push {r3, r4, r5, r6, r7, lr}   
	mov r4, r0                      
	movs r0, #0                     
	mov r5, r1                      
	bl 0x801ac6c <SEGGER_RTT_GetAvailWriteSpace>
	cmp.w r0, r5, lsl #2            
	bcs.n 0x801ada6 <TriceNonBlockingDirectWrite+54>
	ldr r2, [pc, #184] @ (0x801ae3c <TriceNonBlockingDirectWrite+204>)
	ldr r4, [pc, #184] @ (0x801ae40 <TriceNonBlockingDirectWrite+208>)
	ldr r3, [r2, #0]                
	adds r3, #1                     
	str r3, [r2, #0]                
	movs r0, #0                     
	bl 0x801ac6c <SEGGER_RTT_GetAvailWriteSpace>
	rsbs r0, r0, #1024 @ 0x400      
	bmi.n 0x801ada4 <TriceNonBlockingDirectWrite+52>
	ldr r3, [r4, #0]                
	cmp r0, r3                      
	it cc                           
	movcc r0, r3                    
	str r0, [r4, #0]                
	pop {r3, r4, r5, r6, r7, pc}    
	b.n 0x801ada4 <TriceNonBlockingDirectWrite+52>
	movs r0, #0                     
	bl 0x801ac6c <SEGGER_RTT_GetAvailWriteSpace>
	lsls r6, r5, #2                 
	cmp r6, r0                      
	bhi.n 0x801ae1e <TriceNonBlockingDirectWrite+174>
	ldr r7, [pc, #144] @ (0x801ae44 <TriceNonBlockingDirectWrite+212>)
	ldrd r3, r1, [r7, #32]          
	ldr r2, [r7, #28]               
	subs r3, r3, r1                 
	cmp.w r5, r3, lsr #2            
	mov.w lr, r3, lsr #2            
	add.w r0, r2, r1                
	bcc.n 0x801ae22 <TriceNonBlockingDirectWrite+178>
	mov r1, r4                      
	cmp.w lr, #0                    
	beq.n 0x801ade4 <TriceNonBlockingDirectWrite+116>
	add.w r1, r4, lr, lsl #2        
	ldr.w r6, [r4], #4              
	adds r0, #4                     
	cmp r4, r1                      
	str.w r6, [r0, #-4]             
	bne.n 0x801add6 <TriceNonBlockingDirectWrite+102>
	sub.w r5, r5, lr                
	subs r0, r5, #1                 
	cbz r5, 0x801adfc <TriceNonBlockingDirectWrite+140>
	ldr.w r4, [r1], #4              
	subs r0, #1                     
[... 126 lines cut ...]
//! #define TRICE_LEAVE_CRITICAL_SECTION } __set_PRIMASK(primaskstate); }
#define TRICE_LEAVE_CRITICAL_SECTION } __set_PRIMASK(primaskstate); }

//#define TRICE_8_BIT_SUPPORT  1
//#define TRICE_16_BIT_SUPPORT 1
//#define TRICE_32_BIT_SUPPORT 1
//#define TRICE_64_BIT_SUPPORT 1

///////////////////////////////////////////////////////////////////////////////
```

**Expected behavior**
The application should never get blocked by the trice logging output.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

-     OS: ubuntu
-     compiler: arm-none-eabi-gcc
-     Version 12

**Additional context**
I already tried with and without the TRICE_ENTER_CRITICAL_SECTION locks, but no change of behaviour.

initdb, 2025-01-30:
same config with 0.72.5 works fine.

rokath, 2025-01-31:
Am I understand you right, that the problem is inside the target code? So the problem persists, no matter you use the the trice_v0.72.5 tool or the trice_v0.73.1 tool?

I downloaded  trice_linux_amd64.tar.gz v0.72.5 and v0.73.1 and compared the folders, but I could not detect any critical changes. 

Due to a context timeout the v0.73.1 release occurred 3 times in GitHub and in-between I deleted the 2 additional drafts. But I assume they where identical. Maybe you compare  v0.72.5 and v0.73.1 by your own too. 4 eyes are better than 2.

Could it be that you have a buffer size issue and you simply have "luck" with v0.72.5? Just for investigation you could change `#define TRICE_DATA_OFFSET` to `64` and `#define TRICE_SINGLE_MAX_SIZE` to `104` or even smaller values. A typical trice message is 4 to 16 bytes long only. The Trice static buffer needs to hold only one singe Trice message.

I would use `SEGGER_RTT_LOCK()` and `SEGGER_RTT_UNLOCK()` for the critical sections if you use SEGGER RTT anyway. For the investigation, do not use Trice macros inside interrupts or parallel running tasks.

One more thing: Now a newer SEGGER version is used by default. Please set `#define TRICE_DIRECT_SEGGER_RTT_8BIT_WRITE 1` just for investigation. This uses the slower original SEGGER code and we can assume not to have any issue in it.  With `#define TRICE_DIRECT_SEGGER_RTT_32BIT_WRITE 1` a self-written 32-bit code gets active and it relies on an unchanged SEGGER code. Even I could not see any critical changes in the SEGGER update, it is better for investigation to change that back.

initdb, 2025-01-31:
> Am I understand you right, that the problem is inside the target code? So the problem persists, no matter you use the the 
> trice_v0.72.5 tool or the trice_v0.73.1 tool?

yes the tool version doesn't matter, only the difference in the target code.

I will check the other points and report as soon as i get them.

rokath, 2025-01-31:
Could it be, that you do not execute `TriceInit()` before the very first Trice macro?

```C
//! TriceInit needs to run before the first trice macro is executed.
//! Not neseecary for all configurations.
void TriceInit(void) {
#if defined(SEGGER_RTT) && (TRICE_CGO == 0)
	// This is just to force the INIT() call inside SEGGER_RTT.c what allows to use
	// SEGGER_RTT_WriteNoLock or SEGGER_Write_RTT0_NoCheck32 instead of SEGGER_RTT_Write.
	SEGGER_RTT_Write(0, 0, 0); // lint !e534
#endif
```

initdb, 2025-01-31:
the `TriceInit();` in call is the the one before the first log message `TRice(iD(7489), "INFO: APP_THREADX - init...!\n");`. the log gets throw to the trice cli an printed but after this the target code is stuck in a while loop (disassemlby `b.n 0x801ae20 <TriceNonBlockingDirectWrite+176>`). I tried to show this with my code snipped and the debugging arrow (===>).

> I would use SEGGER_RTT_LOCK() and SEGGER_RTT_UNLOCK() for the critical sections if you use SEGGER RTT anyway. For the 
> investigation, do not use Trice macros inside interrupts or parallel running tasks.

i already tried SEGGER_RTT_LOCK() and SEGGER_RTT_UNLOCK(), the gcc way and now lock e.g just the `{` and `}`.

I have to check:
- load symbols to see the c code line
- buffer size as suggested
- `#define TRICE_DIRECT_SEGGER_RTT_8BIT_WRITE 1`

rokath, 2025-01-31:
You could temporarily disable the watchdog for debugging, to find out what condition causes the endless loop. 

You could `#define TRICE_PROTECT 0` just for checking.

rokath, 2025-02-11:
Any results?

initdb, 2025-02-18:
i will have more time to investigate in 3 weeks. for now i have to fix up some other topics.

rokath, 2025-02-22:
One more hint: Check the stack sizes!

rokath, 2025-03-23:
What happens when you switch back to v0.72.5?

rokath, 2025-03-27:
Closing this issue due to inactivity. Please re-open just in case.

closed 2025-03-27 by rokath