rokath/trice #520: Clarify prefix, display and color

KammutierSpule, 2025-03-13:
Can you clarify on trice tool how it displays or not display text/color based on the formats?
eg:
warning vs WARNING vs Warning ?
I noticed some are displayed the text and some dont display.

Others:
```
Warning w wrn warning W WRN WARNING Warn warn WARN 
Info i inf info informal I INF INFO INFORMAL 
Timestamp tim time TIM TIME TIMESTAMP timestamp 
```

rokath, 2025-03-19:
Can we close this?

KammutierSpule, 2025-03-20:
sure, thanks!

but I found my answer only on the go source code, which I understand that, eg "Warning w wrn warning W WRN WARNING Warn warn WARN " is displayed all the same way. Correct?

KammutierSpule, 2025-03-20:
so what are the possibilities? you mentioned two: remove or note remove the "tag".
what about all other possibilities? These are the only two possible behaviors or are there more (eg: for warnings) ?

"Warning w wrn warning W WRN WARNING Warn warn WARN "

rokath, 2025-03-20:
Yes, display tag or not until we have a new idea. If "warnings" should work too, the Go line needs one more entry.

KammutierSpule, 2025-03-21:
ok thank you, I will close it now

closed 2025-03-21 by KammutierSpule