rokath/trice #522: trice did not print for the first time after the program started, and will not print until after the reset.

14kkkkkk, 2025-03-18:
**Describe the bug**
A clear and concise description of what the bug is.
In the direct_output mode +static_buffer mode, the id data in the buffer is manually read and written to the write interface of the uart instead of rtt.

rokath, 2025-03-18:
Please provide your `triceConfig.h` file. I guess, you have enabled `TRICE_UART` too and that is not recommended in direct mode.

14kkkkkk, 2025-03-19:
/*! \file triceConfig.h
\author Thomas.Hoehenleitner [at] seerose.net
*******************************************************************************/

#ifndef TRICE_CONFIG_H_
#define TRICE_CONFIG_H_

#ifdef __cplusplus
extern "C" {
#endif

//! TRICE_CLEAN, if found inside triceConfig.h, is modified by the Trice tool to silent editor warnings in the cleaned state.
#define TRICE_CLEAN 0 // Do not define this at an other place! But you can delete this here.

// hardware specific trice lib settings
// #include "main.h" 
extern uint32_t get_systick(void);
#define TriceStamp16 (get_systick() & 0xffff)    // 0...999 us
#define TriceStamp32 get_systick()  // 0...2^32-1 ms (wraps after 49.7 days)

#define TRICE_BUFFER TRICE_STATIC_BUFFER //TRICE_RING_BUFFER
#define TRICE_DEFERRED_BUFFER_SIZE 2000

// Windows: trice log -p jlink -args "-Device STM32F030R8" -pf none -prefix off -hs off -d16 -showID "deb:%5d" -i ../../demoTIL.json -li ../../demoLI.json
// Unix:   ./RTTLogUnix.sh or manually:
// 		Terminal 1: rm -f ./temp/trice.bin && JLinkRTTLogger -Device STM32G0B1RE -If SWD -Speed 4000 -RTTChannel 0 ./temp/trice.bin
//      Terminal 2: touch ./temp/trice.bin && trice log -p FILE -args ./temp/trice.bin -pf none -prefix off -hs off -d16 -ts ms -i ../../demoTIL.json -li ../../demoLI.json
#define TRICE_DIRECT_OUTPUT 1
#define TRICE_DIRECT_SEGGER_RTT_32BIT_WRITE 0
#define BUFFER_SIZE_UP (2048) // "TRICE_DIRECT_BUFFER_SIZE"

// Windows: trice log -p com3         -prefix off -hs off -pw MySecret -pf COBS -i ../../demoTIL.json -li ../../demoLI.json
// Unix:    trice log -p /dev/ttyACM0 -prefix off -hs off -pw MySecret -pf COBS -i ../../demoTIL.json -li ../../demoLI.json
#define TRICE_DIAGNOSTICS 0
#define TRICE_DEFERRED_OUTPUT 0

#define TRICE_DEFERRED_XTEA_ENCRYPT 0
#define TRICE_DEFERRED_OUT_FRAMING TRICE_FRAMING_NONE//TRICE_FRAMING_COBS
#define TRICE_DEFERRED_UARTA 0
#define TRICE_UARTA 1

extern void ext_int_disable(void);
extern void ext_int_enable(void);
#define TRICE_ENTER_CRITICAL_SECTION do{ext_int_disable();}while(0);
#define TRICE_LEAVE_CRITICAL_SECTION do{ext_int_enable();}while(0);

void TriceHeadLine(char* name);
void LogTriceConfiguration(void);
void SomeExampleTrices(int burstCount);

#define TRICE_FULL_CHECK

#ifdef __cplusplus
}
#endif

#endif /* TRICE_CONFIG_H_ */