rokath/trice #523: trice insert

KammutierSpule, 2025-04-10:
**Describe the bug**
If I forget the %format on trices lines, the trice insert tool will give me confusion error messages.

TRice32("text", var);

**To Reproduce**
Steps to reproduce the behavior:

place multiple Trices on file but in the middle somewhere, forget to put a %d
TRice32("text", var);

**Expected behavior**
The exact line that has the issue

rokath, 2025-04-10:
Please, what is the error message exactly you see?

KammutierSpule, 2025-04-11:
```
// trice insert
// ../main/src/ping.c line 6 format specifier count 0 != parameter count 1 for
// 2025/04/11 13:36:56 1 walk errors
void test_functions(void) {
  int var;
  Trice0("testOK1");  // This line is ok, but "line 6 format specifier count 0
                      // != parameter count 1 for {Trice32 test:}"
  var = 1;            //
  var = 2;            //
  var = 3;            //
  Trice32("test:", var);  // This line is not good, should be line 11 error
  Trice0("testOK2");
}
```

rokath, 2025-04-11:
What I put together:

The error message is correct, despite it shows line number 6 but should show line number 11.

I can have a closer look only in a few days.

rokath, 2025-04-13:
Inside the branch `devel` there is now a fix including a successful but disabled test `_TestInsertIDsForNewTrice_Issue_523`, which is disabled. Please check against your project too.

rokath, 2025-04-25:
Merged into master

closed 2025-04-25 by rokath