rokath/trice #525: Is trice numbering for.inc files not supported now

14kkkkkk, 2025-04-23:
Is trice numbering for.inc files not supported now. I am now trying to replace print_err in the project with trice and then use the upper computer trice for numbering, but I find that it cannot be recognized

rokath, 2025-04-23:
Unfortunately I cannot understand the problem you have. Please provide some details or example data.

rokath, 2025-04-24:
In file Update.go you can find

```Go
	// patSourceFile is a regex pattern matching any source file for patching
	patSourceFile = "(\\.c|\\.h|\\.cc|\\.cpp|\\.hpp)$"
```

You can adapt this line to your needs and compile the Trice tool with `go install ./...` from inside the Trice folder.

If it is really needed we could add `|\\.inc` or other extensions permanently.

Using Trice macros inside include-files would cause a re-translation of all sources including those files **OR** use the `-cache` switch.

14kkkkkk, 2025-04-25:
Thank you for solving this problem

closed 2025-04-25 by rokath

rokath, 2025-04-25:
With release 0.74.0 the `.inc` file extension is supported.