rokath/trice #526: Shadow declaration of triceSingleBufferStartWritePosition in trice.h

joaolehodey, 2025-04-29:
**Description:**

While compiling a project that uses the Trice logging system, the following compiler warning (or error, depending on flags) is raised:

warning: declaration of 'triceSingleBufferStartWritePosition' shadows a global declaration [-Wshadow]
(error: declaration of 'triceSingleBufferStartWritePosition' shadows a global declaration [-Werror=shadow])

File: _trice.h_
Symbol: _triceSingleBufferStartWritePosition_
**Issue:** The variable is declared locally in a scope that conflicts with a global declaration of the same name.

This may cause unexpected behavior or confusion during development, especially with strict compiler flags enabled. It would be helpful to rename the local declaration or refactor to avoid the shadowing.

rokath, 2025-05-02:
Thanks for reporting! Strange, that the `./testAll.sh` script did not detect that. May I ask for your `triceConfig.h` file and your compiler settings? You are on the latest release, right?

joaolehodey, 2025-05-05:
Hello, thank you for your quick reply. 

- The trice config (in zip as issues do not allow .h files): 
[triceConfig.zip](https://github.com/user-attachments/files/20039112/triceConfig.zip)

- Compiller settings: **I believe all you need to cause the error is:** -Wshadow -Werror ,**however here you have the full ones:** -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c++17 -O0 -g -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -march=armv7e-m+fp.dp -mtune=cortex-m7 -mthumb -O2 -g3 -DNDEBUG -Wall -Wextra -Wpedantic -Werror -Wdouble-promotion -Wfloat-equal -Wredundant-decls -Wshadow -Wswitch-default -Wno-packed-bitfield-compat -funsigned-char -ffunction-sections -fdata-sections
- I have tested it using the latest release yes.

Thank you. In case you need anything else, just tell me.

rokath, 2025-05-07:
`-Wshadow` is indeed causing error messages. Please be patient - it will take a while to fix that and, maybe, other warnings as well. My assumption is, that these warnings are not a sign of hidden bugs - but who knows. Inbetween you could compare the `triceConfig.h` files in the examples against your `triceConfig.h` and reduce it to the essentials. Many (default) settings are in the `triceDefaultConfig.h` now, only different settings are needed in the project specific `triceConfig.h` and for example `TRICE_MODE` is not used anymore.

rokath, 2025-05-10:
In branch **issue526shadow** is now a fix. Please check it.

Some background information: 

I was not aware of the `-Wshadow` flag and after adding it to the tests, also some other locally defined variables, like `{ uint32_t* p =...`, triggered this warning and it was no big deal to change the Trice code for them.

The variable `uint32_t* const triceSingleBufferStartWritePosition` is a global variable just for the `TRICE_STATIC_BUFFER` configuration and defined only on the stack for the other configurations. Because of its general global visibility

```C
extern uint32_t* const triceSingleBufferStartWritePosition;
```

the `-Wshadow` warning was triggered for the other configurations.

Simply to adapt that into

```C
#if TRICE_BUFFER == TRICE_STATIC_BUFFER && TRICE_OFF == 0
extern uint32_t* const triceSingleBufferStartWritePosition;
#endif
```

is no solution, because this macro definition in `trice.h` than is not accepted by the compiler in all configurations:

```C
#define TRICE_DIAGNOSTICS_SINGLE_BUFFER_KEEP_START \
	triceSingleBufferStartWritePosition = TriceBufferWritePosition;
``` 

So I decided to have `triceSingleBufferStartWritePosition` as global variable in all configurations, what is possible only without the `const` declaration.

rokath, 2025-05-31:
issue526shadow branch is now merged into main.

closed 2025-06-05 by rokath